open-mmlab/OpenPCDet #303: How to split train/val set for kitti official testing?

YoungZiyu, 2020-10-04:
Hello shaoshuai,

Thanks for your opening source codes, but I still have 1 question. In your paper, PV-RCNN split the 7481 training set into 80% train split + 20% val split, can  you share these txt files? 

thanks for your reply~~~

turboxin, 2020-10-15:
Same question, looking forward to 80%-20% split file. @sshaoshuai 

Besides, would 80-20% split gives better results on test set than 50-50% split？

sshaoshuai, 2020-10-19:
The performance of kitti test split is not stable, and it could not guarantee that more data will achieve better results on KITTI test server. You need to split the training data by yourself if you want to use other split instead of 50%-50%.

closed 2020-10-20 by sshaoshuai