open-mmlab/OpenPCDet #305: help，KeyError: 'point_cloud'

wind19970924, 2020-10-05:
the error in running train.py
File "/home/nvidia/OpenPCDet-master/pcdet/datasets/kitti/kitti_dataset.py", line 347, in __getitem__
    sample_idx = info['point_cloud']['lidar_idx']
KeyError: 'point_cloud'.

thanks a lot

jihanyang, 2020-10-06:
Try to check your created kitti infos. It looks like something wrong for your created infos.

closed 2020-10-19 by sshaoshuai

Rajat-Mehta, 2020-11-22:
@jihanyang what should we check in info files? I am also getting this error? 

@wind19970924 how to resolve this issue?

guzuoxu, 2021-11-08:
> the error in running train.py File "/home/nvidia/OpenPCDet-master/pcdet/datasets/kitti/kitti_dataset.py", line 347, in **getitem** sample_idx = info['point_cloud']['lidar_idx'] KeyError: 'point_cloud'.
> 
> thanks a lot

have you solved this problem?

crjxixixi, 2022-01-06:
> > the error in running train.py File "/home/nvidia/OpenPCDet-master/pcdet/datasets/kitti/kitti_dataset.py", line 347, in **getitem** sample_idx = info['point_cloud']['lidar_idx'] KeyError: 'point_cloud'.
> > thanks a lot
> 
> have you solved this problem?
  have you solved the problem?  I meet the same problem as U