open-mmlab/OpenPCDet #307: evaluation (once again)

MartinHahner88, 2020-10-07:
How can the bold values be different?
Aren't they supposed to be the same or are the outputted, <ins>underlined</ins> IoU thresholds wrong?
I don't get it. @sshaoshuai 

**Car AP_R40@0.70**, 0.70, 0.70:
bbox AP:62.7265, 60.2641, 57.1005
bev  AP:59.4660, 59.3951, 54.7493
**3d   AP:35.0134**, 34.9176, 32.5337
aos  AP:49.82, 46.68, 43.31

**Car AP_R40@<ins>0.70</ins>**, 0.50, 0.50:
bbox AP:62.7265, 60.2641, 57.1005
bev  AP:77.3898, 77.4689, 73.4903
**3d   AP:73.2033**, 72.8325, 68.5751
aos  AP:49.82, 46.68, 43.31

######################################

**Pedestrian AP_R40@0.50**, 0.50, 0.50:
bbox AP:30.8180, 31.9491, 31.3791
bev  AP:12.9555, 14.1526, 14.3572
**3d   AP:9.2486**, 10.0193, 9.6459
aos  AP:16.68, 17.14, 16.99

**Pedestrian AP_R40@<ins>0.50</ins>**, 0.25, 0.25:
bbox AP:30.8180, 31.9491, 31.3791
bev  AP:28.5871, 30.4799, 30.8501
**3d   AP:27.5398**, 29.3073, 29.3918
aos  AP:16.68, 17.14, 16.99

######################################

**Cyclist AP_R40@0.50**, 0.50, 0.50:
bbox AP:29.0840, 27.7680, 27.0108
bev  AP:28.9201, 27.6231, 26.6130
**3d   AP:24.9547**, 24.2983, 22.8519
aos  AP:21.79, 20.73, 20.11

**Cyclist AP_R40@<ins>0.50</ins>**, 0.25, 0.25:
bbox AP:29.0840, 27.7680, 27.0108
bev  AP:30.0657, 28.8583, 27.9719
**3d   AP:30.0657**, 28.8583, 27.9719
aos  AP:21.79, 20.73, 20.11

MartinHahner88, 2020-10-07:
**Car AP_R40@0.70**, 0.70, 0.70:
**bbox AP:62.7265**, 60.2641, 57.1005
bev  AP:59.4660, 59.3951, 54.7493
3d   AP:35.0134, 34.9176, 32.5337
aos  AP:49.82, 46.68, 43.31

**Car AP_R40@<ins>0.70</ins>**, 0.50, 0.50:
**bbox AP:62.7265**, 60.2641, 57.1005
bev  AP:77.3898, 77.4689, 73.4903
3d   AP:73.2033, 72.8325, 68.5751
aos  AP:49.82, 46.68, 43.31

######################################

**Pedestrian AP_R40@0.50**, 0.50, 0.50:
**bbox AP:30.8180**, 31.9491, 31.3791
bev  AP:12.9555, 14.1526, 14.3572
3d   AP:9.2486, 10.0193, 9.6459
aos  AP:16.68, 17.14, 16.99

**Pedestrian AP_R40@<ins>0.50</ins>**, 0.25, 0.25:
**bbox AP:30.8180**, 31.9491, 31.3791
bev  AP:28.5871, 30.4799, 30.8501
3d   AP:27.5398, 29.3073, 29.3918
aos  AP:16.68, 17.14, 16.99

######################################

**Cyclist AP_R40@0.50**, 0.50, 0.50:
**bbox AP:29.0840**, 27.7680, 27.0108
bev  AP:28.9201, 27.6231, 26.6130
3d   AP:24.9547, 24.2983, 22.8519
aos  AP:21.79, 20.73, 20.11

**Cyclist AP_R40@<ins>0.50</ins>**, 0.25, 0.25:
**bbox AP:29.0840**, 27.7680, 27.0108
bev  AP:30.0657, 28.8583, 27.9719
3d   AP:30.0657, 28.8583, 27.9719
aos  AP:21.79, 20.73, 20.11

closed 2020-10-07 by MartinHahner88

paland3, 2021-06-14:
See my answer in #432 

https://github.com/open-mmlab/OpenPCDet/issues/432#issuecomment-860543113