open-mmlab/OpenPCDet #308: Unable to reproduce the results of nuScenes

yeyang1021, 2020-10-08:
I used the default settings to train the nuscenes dataset for 20 epochs.
The results:
2020-10-08 10:55:02,123   INFO  ----------------Nuscene detection_cvpr_2019 results-----------------
***car error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.19, 0.15, 0.08, 0.29, 0.25 | 30.23, 36.51, 39.30, 41.76 | mean AP: 0.3694980287568553
***truck error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.41, 0.21, 0.11, 0.31, 0.26 | 4.47, 11.09, 14.46, 15.67 | mean AP: 0.1142404557716026
***construction_vehicle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.84, 0.49, 1.21, 0.09, 0.48 | 0.00, 0.00, 0.09, 0.45 | mean AP: 0.0013674274124315946
***bus error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.50, 0.19, 0.08, 0.55, 0.19 | 4.98, 14.73, 23.86, 25.58 | mean AP: 0.17285901373045104
***trailer error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.79, 0.22, 0.31, 0.30, 0.28 | 0.00, 0.46, 3.95, 5.42 | mean AP: 0.02460108796236435
***barrier error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.64, 0.29, 0.10, nan, nan | 4.13, 15.49, 24.64, 29.96 | mean AP: 0.1855463415027166
***motorcycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.23, 0.24, 0.47, 0.73, 0.31 | 15.21, 18.29, 18.97, 19.24 | mean AP: 0.17927265983891655
***bicycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.25, 0.31, 0.70, 0.43, 0.09 | 0.15, 0.29, 0.33, 0.37 | mean AP: 0.002861717324087194
***pedestrian error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.46, 0.29, 1.17, 0.73, 0.32 | 3.10, 5.27, 6.85, 9.13 | mean AP: 0.06090067635335221
***traffic_cone error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.51, 0.37, nan, nan, nan | 1.99, 3.25, 5.03, 8.77 | mean AP: 0.04758176893069175
--------------average performance-------------
trans_err:       0.4805
scale_err:       0.2753
orient_err:      0.4688
vel_err:         0.4271
attr_err:        0.2718
mAP:     0.1159
NDS:     0.3656

sshaoshuai, 2020-10-19:
Have you modified the codes or configs? I have trained multiple times and your results are not reasonable.

yeyang1021, 2020-10-20:
> Have you modified the codes or configs? I have trained multiple times and your results are not reasonable.

I downloaded OpenPCDet from github again and tested the provided nuScenes model.
The default configuration is used.

The result is as follows:
mAP: 0.0537
mATE: 0.6625
mASE: 0.6185
mAOE: 0.6631
mAVE: 0.6382
mAAE: 0.5249
NDS: 0.2161
Eval time: 24.8s

Per-class results:
Object Class    AP      ATE     ASE     AOE     AVE     AAE
car     0.299   0.181   0.153   0.084   0.288   0.261
truck   0.057   0.434   0.249   0.203   0.273   0.250
bus     0.089   0.511   0.186   0.357   0.598   0.222
trailer 0.000   1.000   1.000   1.000   1.000   1.000
construction_vehicle    0.000   1.000   1.000   1.000   1.000   1.000
pedestrian      0.000   1.000   1.000   1.000   1.000   1.000
motorcycle      0.091   0.238   0.267   0.512   0.464   0.356
bicycle 0.000   0.261   0.330   0.812   0.482   0.109
traffic_cone    0.000   1.000   1.000   nan     nan     nan
barrier 0.000   1.000   1.000   1.000   nan     nan
2020-10-20 08:26:37,597   INFO  ----------------Nuscene detection_cvpr_2019 results-----------------
***car error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.18, 0.15, 0.08, 0.29, 0.26 | 26.35, 30.23, 31.24, 31.97 | mean AP: 0.29946538313132837
***truck error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.43, 0.25, 0.20, 0.27, 0.25 | 1.74, 4.78, 7.58, 8.83 | mean AP: 0.05732718808144324
***construction_vehicle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, 1.00, 1.00 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***bus error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.51, 0.19, 0.36, 0.60, 0.22 | 2.00, 7.19, 12.74, 13.79 | mean AP: 0.08929415627377697
***trailer error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, 1.00, 1.00 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***barrier error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, nan, nan | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***motorcycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.24, 0.27, 0.51, 0.46, 0.36 | 7.85, 8.97, 9.60, 9.87 | mean AP: 0.09070364497854315
***bicycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.26, 0.33, 0.81, 0.48, 0.11 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***pedestrian error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, 1.00, 1.00 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***traffic_cone error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, nan, nan, nan | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
--------------average performance-------------
trans_err:       0.6625
scale_err:       0.6185
orient_err:      0.6631
vel_err:         0.6382
attr_err:        0.5249
mAP:     0.0537
NDS:     0.2161

MartinHahner, 2020-10-20:
Maybe your nuScenes dataset is corrupted / incomplete.
I was able to reproduce the nuScenes numbers (see below).

[image]

yeyang1021, 2020-10-20:
> Maybe your nuScenes dataset is corrupted / incomplete.
> I was able to reproduce the nuScenes numbers (see below).
> 
> [image]

I also used mmdetection3d to test the nuscenes database and got the correct result.
So, this is not a problem with the dataset.

sshaoshuai, 2020-10-21:
@yeyang1021 Could you get correct numbers with KITTI pretrained models? And what's your version of spconv, pytorch, gcc, cuda?

yeyang1021, 2020-10-25:
> @yeyang1021 Could you get correct numbers with KITTI pretrained models? And what's your version of spconv, pytorch, gcc, cuda?
I did not use the pre-trained KITTI model, but the test results of the model, which trained by myself are normal.
gcc version 7.5.0 
pytorch 1.1
cuda 10.0
spconv 1.0

sshaoshuai, 2020-11-16:
When does you download your nuscene models? 
I have updated the NuScenes models in 2020-08-10 as mentioned in the changelog of README.md.

yeyang1021, 2020-11-17:
> When does you download your nuscene models?
> I have updated the NuScenes models in 2020-08-10 as mentioned in the changelog of README.md.

Thanks for your reply.
Now I can get the correct test result.
I changed the environment.
pytorch 1.3
cuda 10.0
spconv 1.2

sshaoshuai, 2020-11-17:
Glad to hear it.

closed 2020-11-17 by sshaoshuai