open-mmlab/OpenPCDet #311: The question of fix_random_seed

wuyujiji, 2020-10-09:
hi, recently I want to reproduce the provided result of usr's, the usr's result is obtained by training 20 epochs with pv_rcnn.  When I trained 20 epochs and then tested the result, which is not inconsistent with usr's. 
    I think the reason the that not set _fix_random_seed=True_ , so the usr re-train the model with  _fix_random_seed=True_ and get the new result. But the result is also inconsistent with my re-trained results by setting _fix_random_seed=True_.
    Then I check the source code of common_utils.set_random_seed, I found that it is not set torch.cuda.manual_seed(seed). 

`def set_random_seed(seed):
    random.seed(seed)
    np.random.seed(seed)
    torch.manual_seed(seed)
    torch.backends.cudnn.deterministic = True
    torch.backends.cudnn.benchmark = False
`
    Could you please help me analyze that this is the reason to cause the inconsistent result? Thank you very much!!

wuyujiji, 2020-10-09:
Another question is that if I want to use pytorch DDP training, according to [https://pytorch.org/docs/1.2.0/cuda.html#torch.cuda.manual_seed_all](pytorch official), whether I need to set torch.cuda.manual_seed_all(seed) ?

jihanyang, 2020-10-10:
Hello, thank you for your notification. In fact, we have already tried to use `torch.cuda.manual_seed(seed)` and `torch.cuda.manual_seed_all(seed)`, but we still cannot get a consistent result for multi-GPU training. We are not sure whether this problem is related to our platform or not, so if you have any successful attempts on multi-GPU training with fixed random seed, please don't hesitate to let us know. I believe many other users also meet this problem.

wuyujiji, 2020-10-19:
Hello，thanks for your reply. Recently I fixed torch.cuda.manual_seed(seed), the result error of distance=[15, 30, 45, 60, 75] is about 0~2%, but the result error of distance > 75 is above 2%. Futhermore, when I train more times with the same code, software and hardware environments, the result error of distance <= 75 is basically consistent， but the result error of distance > 75 is large. Is this a reasonable phenomenon?

wuyujiji, 2020-10-21:
@sshaoshuai Excuse me! could you please share me how long will you disucess about this question? this problem blocks my project progress！

sshaoshuai, 2020-10-21:
@wuyujiji   I think "needs discussion" means this problem needs more discussions from more users who have some experiences about it, and it doesn't mean that I was those people.
Besides, I have tried to fix the seed for several times and many strategies before, but nothing works.

closed 2022-03-08 by github-actions[bot]

jisoo0-0, 2023-11-13:
same issue..