open-mmlab/OpenPCDet #312: How is possible evaluate models without the training labels?

Rikas98, 2020-10-09:
I have a question about training labels.
I not understand how is possible evaluate models in test split without the training labels.
For example, in this case, isn't it necessary to have the ground truth to compare with the prevision of our model?
Thank you in advance for your help.
PS: Sorry if this question is stupid but I am new in this area.

jihanyang, 2020-10-09:
You can submit the generated results on the leaderboard of dataset, such as kitti, nuscene, lyft, waymo, etc. With your submitted prediction, the online evaluation system will give you the performance result.

MartinHahner, 2020-10-09:
> I have a question about training labels.
> I not understand how is possible evaluate models in test split without the training labels.
> For example, in this case, isn't it necessary to have the ground [...]

The "test" split in this repository is actually the _validation_ split, where labels are publicly available.

Rikas98, 2020-10-09:
@jihanyang  @MartinHahner  Thank you for all your answers.
@jihanyang If I submit the results generated in the online evaluation system, are the results published? Because in my case I'm just trying to replicate the results from the PV-RCNN paper and I don't want my results published.

@MartinHahner But the results of the papers are mostly in the test split, and as my main goal is to replicate the results, so I am more interested in the test split.

jihanyang, 2020-10-10:
Your submitted results would not be public until you set them to be public

Rikas98, 2020-10-10:
@jihanyang Thank you.

Rikas98, 2020-10-13:
One more question, if I want to make a 'new' datataset from the existing dataset (e.g. make the dataset only with the images with labels), it is only necessary to create the folders (training and testing) with the images or it is necessary to change something in the files to create kitti db infos.

zhongzhuonan, 2020-10-18:
> You can submit the generated results on the leaderboard of dataset, such as kitti, nuscene, lyft, waymo, etc. With your submitted prediction, the online evaluation system will give you the performance result.

Hello, I want to use OpenPCDet to eval testing(KITTI) dataset, not val dataset. What are the main changes?  kitti_dataset.yaml or other code? 

Thank you in advance for your help.

sshaoshuai, 2020-10-19:
@zhongzhuonan , you just need to modify the kitti_dataset.yaml for testing on the `test` split and  add `--save_to_file` argument when you run the `python test.py xxxxx`, and it will save the test results automatically.

zhongzhuonan, 2020-10-19:
> @zhongzhuonan , you just need to modify the kitti_dataset.yaml for testing on the `test` split and add `--save_to_file` argument when you run the `python test.py xxxxx`, and it will save the test results automatically.

Thank you very much. I'll try it

closed 2020-10-22 by Rikas98