open-mmlab/OpenPCDet #316: not enough arguments for format string

cuge1995, 2020-10-14:
when run `python train.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --batch_size 2 --epochs 40`

the log is 
```shell
2020-10-14 09:23:28,192   INFO  *************** Performance of EPOCH 30 *****************
2020-10-14 09:23:28,192   INFO  Generate label finished(sec_per_example: 0.0825 second).
2020-10-14 09:23:28,192   INFO  recall_roi_0.3: 0.961499
2020-10-14 09:23:28,192   INFO  recall_rcnn_0.3: 0.962524
2020-10-14 09:23:28,192   INFO  recall_roi_0.5: 0.903463
2020-10-14 09:23:28,192   INFO  recall_rcnn_0.5: 0.915423
2020-10-14 09:23:28,192   INFO  recall_roi_0.7: 0.635209
2020-10-14 09:23:28,192   INFO  recall_rcnn_0.7: 0.708566
2020-10-14 09:23:28,196   INFO  Average predicted number of objects(3769 samples): 11.723
Traceback (most recent call last):
  File "train.py", line 198, in <module>
    main()
  File "train.py", line 191, in main
    dist_test=dist_train
  File "/home/amax/OpenPCDet/tools/test.py", line 121, in repeat_eval_ckpt
    result_dir=cur_result_dir, save_to_file=args.save_to_file
  File "/home/amax/OpenPCDet/tools/eval_utils/eval_utils.py", line 113, in eval_one_epoch
    output_path=final_output_dir
  File "/home/amax/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 329, in evaluation
    from .kitti_object_eval_python import eval as kitti_eval
  File "/home/amax/OpenPCDet/pcdet/datasets/kitti/kitti_object_eval_python/eval.py", line 6, in <module>
    from .rotate_iou import rotate_iou_gpu_eval
  File "/home/amax/OpenPCDet/pcdet/datasets/kitti/kitti_object_eval_python/rotate_iou.py", line 263, in <module>
    def rotate_iou_kernel_eval(N, K, dev_boxes, dev_query_boxes, dev_iou, criterion=-1):
  File "/home/amax/anaconda3/envs/pcdetnew/lib/python3.6/site-packages/numba-0.51.2-py3.6-linux-x86_64.egg/numba/cuda/decorators.py", line 136, in kernel_jit
    return Dispatcher(func, sigs, bind=bind, targetoptions=targetoptions)
  File "/home/amax/anaconda3/envs/pcdetnew/lib/python3.6/site-packages/numba-0.51.2-py3.6-linux-x86_64.egg/numba/cuda/compiler.py", line 808, in __init__
    self.compile(sigs[0])
  File "/home/amax/anaconda3/envs/pcdetnew/lib/python3.6/site-packages/numba-0.51.2-py3.6-linux-x86_64.egg/numba/cuda/compiler.py", line 935, in compile
    kernel.bind()
  File "/home/amax/anaconda3/envs/pcdetnew/lib/python3.6/site-packages/numba-0.51.2-py3.6-linux-x86_64.egg/numba/cuda/compiler.py", line 576, in bind
    self._func.get()
  File "/home/amax/anaconda3/envs/pcdetnew/lib/python3.6/site-packages/numba-0.51.2-py3.6-linux-x86_64.egg/numba/cuda/compiler.py", line 446, in get
    ptx = self.ptx.get()
  File "/home/amax/anaconda3/envs/pcdetnew/lib/python3.6/site-packages/numba-0.51.2-py3.6-linux-x86_64.egg/numba/cuda/compiler.py", line 414, in get
    arch = nvvm.get_arch_option(*cc)
  File "/home/amax/anaconda3/envs/pcdetnew/lib/python3.6/site-packages/numba-0.51.2-py3.6-linux-x86_64.egg/numba/cuda/cudadrv/nvvm.py", line 345, in get_arch_option
    return 'compute_%d%d' % arch
TypeError: not enough arguments for format string
```

sshaoshuai, 2020-11-07:
Have you solved the problem? It seems like a enviroment problem.

sshaoshuai, 2020-11-09:
Feel free to re-open it if you still have problems.

closed 2020-11-09 by sshaoshuai