open-mmlab/OpenPCDet #320: kiti dataset info

suvasis, 2020-10-16:
hi,

New to this project, followed all the documentation. Running into the following issue:

What is missing here?

Generate the data infos by running the following command:
python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml

test sample_idx: 007512
test sample_idx: 007513
test sample_idx: 007514
test sample_idx: 007515
test sample_idx: 007516
test sample_idx: 007517
Traceback (most recent call last):
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/projectdata/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 434, in <module>
    create_kitti_infos(
  File "/projectdata/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 414, in create_kitti_infos
    kitti_infos_test = dataset.get_infos(num_workers=workers, has_label=False, count_inside_pts=False)
  File "/projectdata/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 191, in get_infos
    return list(infos)
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/concurrent/futures/_base.py", line 611, in result_iterator
    yield fs.pop().result()
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/concurrent/futures/_base.py", line 432, in result
    return self.__get_result()
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/concurrent/futures/_base.py", line 388, in __get_result
    raise self._exception
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/projectdata/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 128, in process_single_scene
    image_info = {'image_idx': sample_idx, 'image_shape': self.get_image_shape(sample_idx)}
  File "/projectdata/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 69, in get_image_shape
    assert img_file.exists()
AssertionError

jihanyang, 2020-10-16:
Hello, 
As the error raise, you might miss some files or give incorrect image path. Please try to check what the `image_file` is when program raises this error.

suvasis, 2020-10-16:
Got it. Thanks for your help Jihang.

closed 2020-10-19 by sshaoshuai

Manueljohnson063, 2021-03-26:
@sshaoshuai Hai could you please tell how you solved it ,currently i am stuck with it!!!

suvasis, 2021-03-26:
> @sshaoshuai Hai could you please tell how you solved it ,currently i am stuck with it!!!

Check the image files. There are some missing one, either delete those or move it somewhere. Check the .txt and actual image files there are differences. They have to be in sync.

Manueljohnson063, 2021-03-27:
Hi @suvasis ,
Thanks for the replay , actually i had put four folder in the training (calib,image_2,label_2,velodyne) ,and in each folder i have 2 files (000000 and 000001) each .Same in the case of testing folder(calib,image_2,velodyne) with (000002 and 000003) .also in the imageSets folder i have three text files train.txt(000000,000001) and in val.txt (000002 and 000003) in test.txt( (000000 , 000001,000002,000003)

but output is

/usr/lib/python3.7/runpy.py:125: RuntimeWarning: 'pcdet.datasets.kitti.kitti_dataset' found in sys.modules after import of package 'pcdet.datasets.kitti', but prior to execution of 'pcdet.datasets.kitti.kitti_dataset'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
---------------Start to generate data infos---------------
train sample_idx: 000000
train sample_idx: 000001
Kitti info train file is saved to /content/drive/MyDrive/OpenPCDet/data/kitti/kitti_infos_train.pkl
val sample_idx: 000002
val sample_idx: 000003
Traceback (most recent call last):
File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/content/drive/MyDrive/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 439, in <module>
    save_path=ROOT_DIR / 'data' / 'kitti'
  File "/content/drive/MyDrive/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 404, in create_kitti_infos
    kitti_infos_val = dataset.get_infos(num_workers=workers, has_label=True, count_inside_pts=True)
  File "/content/drive/MyDrive/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 192, in get_infos
    return list(infos)
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 598, in result_iterator
    yield fs.pop().result()
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 428, in result
    return self.__get_result()
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/content/drive/MyDrive/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 129, in process_single_scene
    image_info = {'image_idx': sample_idx, 'image_shape': self.get_image_shape(sample_idx)}
  File "/content/drive/MyDrive/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 69, in get_image_shape
    assert img_file.exists()
AssertionError

Manueljohnson063, 2021-03-31:
@suvasis Sir could you please tell what change you have made.i am stuck with this.

thiesymon, 2023-11-02:
solution for this question is to sign in the website of [kitti_datasets](https://www.cvlibs.net/datasets/kitti/eval_object.php?obj_benchmark=3d) to download four zip files : data_object_label_2.zip data_object_calib.zip data_object_velodyne.zip data_object_image_2.zip to the directory OpenPCDet/data/kitti/ and unzip those zip files would work !

Sun-Cold, 2024-06-14:
如果你在windows上用WSL，并且数据集都放在windows目录下，可以参考我的方法：在代码目录下创建一个指向数据集的软链接（假设数据集在_D:\CodeField\dataset\kitti_）：
`ln -s /mnt/d/CodeField/dataset/kitti kitti`

If you are using WSL on windows and the dataset is in the windows directory, you can refer to my method: create a soft link to the dataset in the code directory (assuming the dataset is in _D:\CodeField\dataset\kitti_): 
`ln -s /mnt/d/CodeField/dataset /kitti kitti`