open-mmlab/OpenPCDet #321: segmentation fault(core dumped)

Dir-b, 2020-10-16:
When using the demo.py, I encountered a problem——segmentation fault(core dumped).
After debugging，i found the problem:line 20 in the detector3d_template.py, "self.register_buffer('global_step', torch.LongTensor(1).zero_())".
I'm stuck with this question,can you help me...thanks...

sshaoshuai, 2020-10-19:
Could you please post your environment (cuda, gcc, pytorch, spconv) and the build logs?

Dir-b, 2020-10-19:
I have found the cause of the problem.
In other issues,it was mentioned.he said"mayavi depends on pyqt4."So,i tried to uninstall pyqt5 in my environmental. It worked!!!
But I don't know what this has to do with where the program crashes.

closed 2020-10-19 by sshaoshuai

RSSQ2333, 2022-01-13:
I have downgraded the open3d from 0.14 to 0.13. Then it works!
What's more, some information about my environment is showed as follows:
Ubuntu 18.04
python 3.8
pytorch 1.9.0
cuda 11.3
cudnn 9.0
spconv 2.1