open-mmlab/OpenPCDet #322: limits of cuda hash failure

suvasis, 2020-10-17:
hi,

I am trining as per the documentation, 
python train.py --cfg cfgs/kitti_models/pv_rcnn.yaml --batch_size 64 

however running into the following issue. I tried various batch sizes, 256, 64, 512 and it hasn't helped. How do I fix this?

I have single GPU AWS instance, g3.4xlarge.

2020-10-17 02:08:43,880   INFO  **********************Start training kitti_models/pv_rcnn(default)**********************
epochs:   0%|                                                                            | 0/80 [04:03<?, ?it/s]
Traceback (most recent call last):                                                        | 0/8 [00:00<?, ?it/s]
  File "train.py", line 198, in <module>
    main()
  File "train.py", line 153, in main
    train_model(
  File "/projectdata/OpenPCDet/tools/train_utils/train_utils.py", line 86, in train_model
    accumulated_iter = train_one_epoch(
  File "/projectdata/OpenPCDet/tools/train_utils/train_utils.py", line 38, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/projectdata/OpenPCDet/pcdet/models/__init__.py", line 30, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/projectdata/OpenPCDet/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/projectdata/OpenPCDet/pcdet/models/backbones_3d/spconv_backbone.py", line 139, in forward
    x = self.conv_input(input_sp_tensor)
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/site-packages/spconv/modules.py", line 134, in forward
    input = module(input)
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/site-packages/spconv/conv.py", line 169, in forward
    outids, indice_pairs, indice_pair_num = ops.get_indice_pairs(
  File "/projectdata/anaconda3/envs/apple2/lib/python3.8/site-packages/spconv/ops.py", line 91, in get_indice_pairs
    res = torch.ops.spconv.get_indice_pairs(indices, batch_size, out_shape,
RuntimeError: /projectdata/spconv/src/spconv/spconv_ops.cc 40
batchSize * outputVolume < std::numeric_limits<int>::max() assert faild. due to limits of cuda hash, the volume of dense space include batch size must less than std::numeric_limits<int>::max() = 2e9

suvasis, 2020-10-17:
hi,
If the BATCH_SIZE is 1, it works. It does not work for 2,4,32... 
The training takes ages and GPU AWS instance, g3.4xlarge, cost $2.5/hr. Is there a work around for this?

sshaoshuai, 2020-10-19:
Maybe this thread could be useful: https://github.com/open-mmlab/OpenPCDet/issues/299

closed 2020-10-24 by sshaoshuai

baraujo98, 2021-04-12:
@suvasis did you find the reason for this error? I'm struggling with the same thing. Maybe its got to do with the VRAM capacity not being enough for those batch sizes...