open-mmlab/OpenPCDet #325: 'gcc' failed with exit status 1

snslmi, 2020-10-20:
when i run    python setup.py develop   the gcc related error occurs, 
i tried to reinstall gcc but still didnt work.

running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.pointnet2.pointnet2_stack.pointnet2_stack_cuda' extension
gcc -pthread -B /home/lotte_fashion_ai/anaconda3/envs/OpenPCDet/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 
-Wall -Wstrict-prototypes -fPIC -I/home/lotte_fashion_ai/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/torch/include -I/home/lot
te_fashion_ai/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/lotte_fashion_ai/anacon
da3/envs/OpenPCDet/lib/python3.7/site-packages/torch/include/TH -I/home/lotte_fashion_ai/anaconda3/envs/OpenPCDet/lib/python3.7/site-pa
ckages/torch/include/THC -I/usr/local/cuda/include -I/home/lotte_fashion_ai/anaconda3/envs/OpenPCDet/include/python3.7m -c pcdet/ops/po
intnet2/pointnet2_stack/src/pointnet2_api.cpp -o build/temp.linux-x86_64-3.7/pcdet/ops/pointnet2/pointnet2_stack/src/pointnet2_api.o -D
TORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=pointnet2_stack_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
In file included from pcdet/ops/pointnet2/pointnet2_stack/src/sampling_gpu.h:5:0,
                 from pcdet/ops/pointnet2/pointnet2_stack/src/pointnet2_api.cpp:6:
/home/lotte_fashion_ai/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/torch/include/ATen/cuda/CUDAContext.h:13:10: fatal error: c
ublas_v2.h: No such file or directory
 #include <cublas_v2.h>
          ^~~~~~~~~~~~~
compilation terminated.
error: command 'gcc' failed with exit status 1

how can i fix this? please help

MartinHahner, 2020-10-20:
What is your gcc version?

`gcc --version`

snslmi, 2020-10-20:
> What is your gcc version?
> 
> `gcc --version`

gcc version is 7.5.0
and i solved this by reinstalling cuda toolkit
thanks anyway!:)

sudo apt-get install nvidia-cuda-toolkit

closed 2020-10-20 by snslmi