open-mmlab/OpenPCDet #329: RuntimeError: can't start new thread

cqsnove, 2020-10-24:
The following error occurred while training:

Traceback (most recent call last):
  File "train.py", line 200, in <module>
    main()
  File "train.py", line 172, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/mnt/disk/home1/cqs/Deformable-PV-RCNN/OpenPCDet/tools/train_utils/train_utils.py", line 93, in train_model
    dataloader_iter=dataloader_iter
  File "/mnt/disk/home1/cqs/Deformable-PV-RCNN/OpenPCDet/tools/train_utils/train_utils.py", line 12, in train_one_epoch
    dataloader_iter = iter(train_loader)
  File "/mnt/disk/home1/cqs/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 279, in __iter__
    return _MultiProcessingDataLoaderIter(self)
  File "/mnt/disk/home1/cqs/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 733, in __init__
    pin_memory_thread.start()
  File "/mnt/disk/home1/cqs/anaconda3/envs/OpenPCDet/lib/python3.7/threading.py", line 852, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread

And  another error occurred while testing:
 libgomp: Thread creation failed: Resource temporarily unavailable

Can anyone help me with this problem?

sshaoshuai, 2020-11-16:
Have you solved the problem? It seems like a problem of environment.

sshaoshuai, 2020-11-22:
Feel free to re-open it if you still have problems.

closed 2020-11-22 by sshaoshuai