open-mmlab/OpenPCDet #333: Testing with custom data in Pcap format

arxidinakbar, 2020-10-29:
Hi, firstly appreciate for this Open-lab, much inspirational it is.
And, I just wondering does anybody had tried to testing with 'Pcap' format dataset, which is collect with Lidar directly(Velodyne product).
I hope get some tips to work it out, thank you!

sharma-n, 2020-11-03:
The easiest method I have figured out is to convert PCAP data to a 4D array using MATLAB: [Reading point cloud data in MATLAB](https://www.mathworks.com/help/vision/ref/velodynefilereader.html). MATLAB is able to split the PCAP data into multiple frames, and process them. You can then export the 4D array into a bin file.

If you don't have MATLAB, I have not found any other easy solution. Most probably need to parse the information from the packets. If you do find something else (which is free unlike MATLAB) then please share!

arxidinakbar, 2020-11-03:
@sharma-n  Thank you so much for reply. 
This is a breakthrough!  I didn't know this function before, unfortunately, I can't afford MATLAB 2020 now. But I would look into this!
I found Pcap file could unpack according to its UDP protocal, yet no more substantial progress.
Hope I could get some idea to tackle this issue, and definitely would update if any progress.

neisakutou, 2020-11-24:
> The easiest method I have figured out is to convert PCAP data to an array using MATLAB: [Reading point cloud data in MATLAB](https://www.mathworks.com/help/vision/ref/velodynefilereader.html). MATLAB is able to split the PCAP data into multiple frames, and process them. You can then export the array into a bin file.
> 
> 
> 
> If you don't have MATLAB, I have not found any other easy solution. Most probably need to parse the information from the packets. If you do find something else (which is free unlike MATLAB) then please share!

Hi all,

Actually, I am running demo on my own dataset which is collected from stationed sensors on vehicle.

Basically, the initial data is pcap packet and it is parsed into single frame(PCD) then.

//todo

closed 2022-03-08 by github-actions[bot]