open-mmlab/OpenPCDet #334: How to use cross-validation

Wjsonr, 2020-10-30:
Hello,
I have a question about the training and verification process of PV-RCNN。
PV-RCNN in OpenPCDet is trained according to the fixed segmentation of train split (3712 samples) and val split (3769 samples)。
I want to implement the "80%+20%" training method mentioned in the original "the model is trained with 80% of all available train+val data and the remaining 20% data is used for validation" and use it on this basis 10-fold cross validation。Do I have to split the 7481 samples by myself and modify the code such as train.py?
Another question is whether K-fold cross validation is used for training before PV-RCNN is submitted to kitti's official website？

sshaoshuai, 2020-11-03:
Yes, you need to split the training data by youself if you would like to use other split strategies instead of 50%-50%. 
I didn't use the K-fold cross validation for submitting to the test set.

closed 2020-11-04 by Wjsonr

Biniang5510, 2021-01-26:
Hi @Wjsonr, did you successfully apply cross-validation?

ZhangYu1ing, 2021-06-10:
> Hello,
> I have a question about the training and verification process of PV-RCNN。
> PV-RCNN in OpenPCDet is trained according to the fixed segmentation of train split (3712 samples) and val split (3769 [...]

Hello, may I ask where is the train set splitter? in which file?
Thanks in advance~