open-mmlab/OpenPCDet #336: Confused about the batch_size for PV-RCNN.

Ultra11nstinct, 2020-11-03:
Hi, @sshaoshuai 
Thanks for your great work.

I recognized that the batch_size used in the paper is 24 with 8 1080ti GPUs for KITTI dataset training.
but i found that for a 1080ti GPU with 11GB memory, it only can handle a batch of 2 which means that 8 1080ti GPUs can only support batch of 16. it doesn't match the number in the paper(24).

Did i ignore something, or it is just a text error?

Looking forward to your reply.

sshaoshuai, 2020-11-03:
Maybe OpenPCDet consumes slightly more GPU-memory than my previous version, since it records lots of tensors in the forward data dict. You just need to use batch size 2 since it could also achieve similar performance with batch size 3 .

Ultra11nstinct, 2020-11-04:
> Maybe OpenPCDet consumes slightly more GPU-memory than my previous version, since it records lots of tensors in the forward data dict. You just need to use batch size 2 since it could also achieve similar performance with batch size 3 .

Thank you, i got it.
I only have 1 2080ti GPU, so i need to ajust the lr on the basis of actual batch size.

closed 2020-11-04 by sshaoshuai