open-mmlab/OpenPCDet #338: PartA2 KITTI training error: cuda execution failed with error 700 an illegal memory access was encountered

sharma-n, 2020-11-03:
Hi,
First of all, thanks for sharing this work the modularization of the code is extremely helpful. I have been trying to train the PartA2 model but it leads to the following error being thrown within the first 50 batches (not always the same iteration):

```shell
Traceback (most recent call last):
  File "OpenPCDet/pcdet/models/roi_heads/partA2_head.py", line 200, in forward
    x_part = self.conv_part(part_features)
  File "/opt/conda/lib/python3.6/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/opt/conda/lib/python3.6/site-packages/spconv/modules.py", line 134, in forward
    input = module(input)
  File "/opt/conda/lib/python3.6/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/opt/conda/lib/python3.6/site-packages/spconv/modules.py", line 134, in forward
    input = module(input)
  File "/opt/conda/lib/python3.6/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/opt/conda/lib/python3.6/site-packages/spconv/conv.py", line 181, in forward
    use_hash=self.use_hash)
  File "/opt/conda/lib/python3.6/site-packages/spconv/ops.py", line 95, in get_indice_pairs
    int(use_hash))
RuntimeError: /OpenPCDet/spconv/src/spconv/indice.cu 274
cuda execution failed with error 700 an illegal memory access was encountered
prepareSubMGridKernel failed
```

I got this error by running the following command:
`CUDA_LAUNCH_BLOCKING=1 python train.py --cfg_file cfgs/kitti_models/PartA2.yaml`

The `CUDA_LAUNCH_BLOCKING=1` was suggested somewhere else to better identify the source of the problem. I am using `spconv v1.2` with `pytorch 1.7.0`. Any ideas?

sshaoshuai, 2020-11-04:
Have you tried to train other models like SECOND and PointPillar?  
Which dataset does you use?

sharma-n, 2020-11-05:
@sshaoshuai I used the KITTI dataset. I also tried training the PV-RCNN model which worked perfectly giving me the expected results.

sshaoshuai, 2020-11-07:
Sorry, I do not know why it happens. Maybe you could try to use smaller batch_size like 3 for training with PartA2.

closed 2020-11-22 by sshaoshuai

ArtificialCleverness, 2021-02-04:
Have you guys found any solution? I have exactly the same issue with same environment. I also noticed that any batch size greater than one can cause this issue, but no prob with batch size 1.

MartinHahner, 2021-05-06:
I believe that this error is strictly related to **spconv v1.2**.
I recently upgraded my environment to use `spconv v1.2` instead of `spconv v1.0` and it worked fine before.

When I try to train PartA2 with batch size 1 it unfortunately does not "fix" the issue for me, instead I get another error (see below).

I am using
- boost 1.73.0
- cmake 3.16.5
- cuda 11.0.3
- cudnn 8.0.5
- gcc 6.3.0
- nccl 2.7.8-1
- python 3.8.5

My output of `pip list` is added below.

```python
2021-05-06 14:46:13,241   INFO  CUDA_VISIBLE_DEVICES=0
2021-05-06 14:46:13,249   INFO  batch_size       1
2021-05-06 14:46:13,249   INFO  epochs           80
2021-05-06 14:46:13,249   INFO  workers          4
2021-05-06 14:46:13,249   INFO  ckpt             None
2021-05-06 14:46:13,250   INFO  pretrained_model None
2021-05-06 14:46:13,250   INFO  launcher         none
2021-05-06 14:46:13,250   INFO  tcp_port         18888
2021-05-06 14:46:13,250   INFO  sync_bn          False
2021-05-06 14:46:13,250   INFO  fix_random_seed  False
2021-05-06 14:46:13,250   INFO  ckpt_save_interval 1
2021-05-06 14:46:13,250   INFO  local_rank       0
2021-05-06 14:46:13,250   INFO  max_ckpt_save_num 80
2021-05-06 14:46:13,250   INFO  merge_all_iters_to_one_epoch False
2021-05-06 14:46:13,251   INFO  set_cfgs         None
2021-05-06 14:46:13,251   INFO  max_waiting_mins 0
2021-05-06 14:46:13,251   INFO  start_epoch      0
2021-05-06 14:46:13,251   INFO  save_to_file     False
2021-05-06 14:46:13,251   INFO  log_level        info
2021-05-06 14:46:13,251   INFO  cfg.LOCAL_RANK: 0
2021-05-06 14:46:13,252   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2021-05-06 14:46:13,252   INFO
cfg.MODEL = edict()
2021-05-06 14:46:13,255   INFO  cfg.MODEL.NAME: PartA2Net
2021-05-06 14:46:13,255   INFO
cfg.MODEL.VFE = edict()
2021-05-06 14:46:13,255   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2021-05-06 14:46:13,255   INFO
cfg.MODEL.BACKBONE_3D = edict()
2021-05-06 14:46:13,256   INFO  cfg.MODEL.BACKBONE_3D.NAME: UNetV2
2021-05-06 14:46:13,256   INFO
cfg.MODEL.MAP_TO_BEV = edict()
2021-05-06 14:46:13,256   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2021-05-06 14:46:13,256   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2021-05-06 14:46:13,256   INFO
cfg.MODEL.BACKBONE_2D = edict()
2021-05-06 14:46:13,256   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
2021-05-06 14:46:13,256   INFO  cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [5, 5]
2021-05-06 14:46:13,256   INFO  cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [1, 2]
2021-05-06 14:46:13,257   INFO  cfg.MODEL.BACKBONE_2D.NUM_FILTERS: [128, 256]
2021-05-06 14:46:13,257   INFO  cfg.MODEL.BACKBONE_2D.UPSAMPLE_STRIDES: [1, 2]
2021-05-06 14:46:13,257   INFO  cfg.MODEL.BACKBONE_2D.NUM_UPSAMPLE_FILTERS: [256, 256]
2021-05-06 14:46:13,257   INFO
cfg.MODEL.DENSE_HEAD = edict()
2021-05-06 14:46:13,257   INFO  cfg.MODEL.DENSE_HEAD.NAME: AnchorHeadSingle
2021-05-06 14:46:13,257   INFO  cfg.MODEL.DENSE_HEAD.CLASS_AGNOSTIC: False
2021-05-06 14:46:13,257   INFO  cfg.MODEL.DENSE_HEAD.USE_DIRECTION_CLASSIFIER: True
2021-05-06 14:46:13,257   INFO  cfg.MODEL.DENSE_HEAD.DIR_OFFSET: 0.78539
2021-05-06 14:46:13,257   INFO  cfg.MODEL.DENSE_HEAD.DIR_LIMIT_OFFSET: 0.0
2021-05-06 14:46:13,258   INFO  cfg.MODEL.DENSE_HEAD.NUM_DIR_BINS: 2
2021-05-06 14:46:13,258   INFO  cfg.MODEL.DENSE_HEAD.ANCHOR_GENERATOR_CONFIG: [{'class_name': 'Car', 'anchor_sizes': [[3.9, 1.6, 1.56]], 'anchor_rotations': [0, 1.57], 'anchor_bottom_heights': [-1.78], 'align_center': False, 'feature_map_stride': 8, 'matched_threshold': 0.6, 'unmatched_threshold': 0.45}, {'class_name': 'Pedestrian', 'anchor_sizes': [[0.8, 0.6, 1.73]], 'anchor_rotations': [0, 1.57], 'anchor_bottom_heights': [-1.78], 'align_center': False, 'feature_map_stride': 8, 'matched_threshold': 0.5, 'unmatched_threshold': 0.35}, {'class_name': 'Cyclist', 'anchor_sizes': [[1.76, 0.6, 1.73]], 'anchor_rotations': [0, 1.57], 'anchor_bottom_heights': [-1.78], 'align_center': False, 'feature_map_stride': 8, 'matched_threshold': 0.5, 'unmatched_threshold': 0.35}]
2021-05-06 14:46:13,258   INFO
cfg.MODEL.DENSE_HEAD.TARGET_ASSIGNER_CONFIG = edict()
[... 916 lines cut ...]
typing-extensions               3.7.4.3
urllib3                         1.26.4
utm                             0.7.0
wandb                           0.10.29
waymo-open-dataset-tf-2-4-0     1.2.0
wcwidth                         0.2.5
webencodings                    0.5.1
Werkzeug                        1.0.1
wheel                           0.35.1
widgetsnbextension              3.5.1
wntr                            0.3.0
wrapt                           1.12.1
xarray                          0.17.0
xlrd                            2.0.1
XlsxWriter                      1.4.0
xlwt                            1.3.0
xmlrunner                       1.7.7
xxhash                          2.0.0
yarl                            1.6.3
zict                            2.0.0
zipp                            3.3.0
zope.event                      4.5.0
zope.interface                  5.4.0

```

MartinHahner, 2021-05-18:
Downgraded to 
- torch 1.4
- cuda 10.1.243
- cudnn 7.6.4
- spconv 1.0

Also see [here](https://github.com/traveller59/spconv/issues/78#issuecomment-631599062) and [there](https://github.com/traveller59/spconv/issues/12#issuecomment-457635066).
Now the training of PartA2 works again.

xuxiaoxxxx, 2022-01-23:
> 

my environment is the same as yours, but I still encount this problem.