open-mmlab/OpenPCDet #341: RuntimeWarning: invalid value encountered in add   obj_points[:, :3] += info['box3d_lidar'][:3]

yustaub, 2020-11-04:
I have transform my custom dataset to kitti format, and when I run `python train.py --cfg-file cfgs/kitti_models/pointpillar.yaml`, and my datsset only have class **Pedestrian**, it can train but there are some **warning**:

`/home/joseph/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py:136: RuntimeWarning: invalid value encountered in add                                                                       | 0/3494 [00:00<?, ?it/s]
  obj_points[:, :3] += info['box3d_lidar'][:3]
/home/joseph/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py:136: RuntimeWarning: invalid value encountered in add
  obj_points[:, :3] += info['box3d_lidar'][:3]
/home/joseph/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py:136: RuntimeWarning: invalid value encountered in add
  obj_points[:, :3] += info['box3d_lidar'][:3]
/home/joseph/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py:136: RuntimeWarning: invalid value encountered in add
  obj_points[:, :3] += info['box3d_lidar'][:3]
/home/joseph/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py:136: RuntimeWarning: invalid value encountered in add
  obj_points[:, :3] += info['box3d_lidar'][:3]
/home/joseph/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py:136: RuntimeWarning: invalid value encountered in add
  obj_points[:, :3] += info['box3d_lidar'][:3]
/home/joseph/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py:136: RuntimeWarning: invalid value encountered in add
  obj_points[:, :3] += info['box3d_lidar'][:3]
/home/joseph/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py:136: RuntimeWarning: invalid value encountered in add
  obj_points[:, :3] += info['box3d_lidar'][:3]
/home/joseph/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:76: RuntimeWarning: overflow encountered in multiply
  points[:, :3] *= noise_scale
/home/joseph/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:76: RuntimeWarning: overflow encountered in multiply
  points[:, :3] *= noise_scale
epochs:   1%|\u2588\u2588\u258e                                                                                                                                                                   | 1/72 [16:58<19:56:10, 1010.86s/it, loss=1.87, lr=0.000307]/home/yuwentao/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:76: RuntimeWarning: overflow encountered in multiply                                                 | 22/3494 [00:06<16:52,  3.43it/s, total_it=3540]
  points[:, :3] *= noise_scale
epochs:   1%|\u2588\u2588\u258e                                                                                                                                                                   | 1/72 [16:58<19:56:10, 1010.86s/it, loss=1.81, lr=0.000307]/home/yuwentao/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:76: RuntimeWarning: overflow encountered in multiply                                                 | 23/3494 [00:07<16:47,  3.45it/s, total_it=3541]
  points[:, :3] *= noise_scale
epochs:   1%|\u2588\u2588\u258e                                                                                                                                                                   | 1/72 [16:58<19:56:10, 1010.86s/it, loss=1.75, lr=0.000307]/home/yuwentao/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:76: RuntimeWarning: overflow encountered in multiply                                                 | 24/3494 [00:07<16:52,  3.43it/s, total_it=3542]
  points[:, :3] *= noise_scale
epochs:   1%|\u2588\u2588\u258e                                                                                                                                                                   | 1/72 [16:59<19:56:10, 1010.86s/it, loss=1.96, lr=0.000307]/home/yuwentao/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:76: RuntimeWarning: overflow encountered in multiply                                                 | 25/3494 [00:07<16:48,  3.44it/s, total_it=3543]
  points[:, :3] *= noise_scale
epochs:   1%|\u2588\u2588\u258e                                                                                                                                                                   | 1/72 [16:59<19:56:10, 1010.86s/it, loss=1.92, lr=0.000307]/home/yuwentao/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:76: RuntimeWarning: overflow encountered in multiply                                                 | 27/3494 [00:08<17:55,  3.22it/s, total_it=3545]
  points[:, :3] *= noise_scale
epochs:   1%|\u2588\u2588\u258e                                                                                                                                                                   | 1/72 [17:00<19:56:10, 1010.86s/it, loss=1.96, lr=0.000307`

does this matters? Very appreciate for your reply!

sshaoshuai, 2020-11-04:
Could you print out the data before and after the warning line?  It seems your data has some problems.

yustaub, 2020-11-04:
`pre:  [[-6.0979257e+00  5.4513931e-01  1.5528565e+31 -1.3553753e+00]
 [ 3.0442011e+19  1.4633905e+19  1.9376342e-01  1.8750000e+00]
 [-8.3080258e+16 -3.9937853e+16 -3.3096598e+10 -1.4266496e+00]
 ...
 [ 4.2872710e+00 -4.8845239e+00 -7.0371407e-01  1.0000000e+00]
 [ 4.2894554e+00 -4.8904548e+00 -4.7602645e-01  1.0000000e+00]
 [ 4.1972418e+00 -4.7903767e+00 -2.4628183e-01  1.0000000e+00]]
/home/yuwentao/project/3D_tracking/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:77: RuntimeWarning: overflow encountered in multiply
  points[:, :3] *= noise_scale
after:  [[-6.1949186e+00  5.5381024e-01  1.5775561e+31 -1.3553753e+00]
 [ 3.0926219e+19  1.4866669e+19  1.9684540e-01  1.8750000e+00]
 [-8.4401725e+16 -4.0573100e+16 -3.3623028e+10 -1.4266496e+00]
 ...
 [ 4.3554635e+00 -4.9622164e+00 -7.1490723e-01  1.0000000e+00]
 [ 4.3576827e+00 -4.9682417e+00 -4.8359805e-01  1.0000000e+00]
 [ 4.2640023e+00 -4.8665719e+00 -2.5019917e-01  1.0000000e+00]]
`
pre means before the warning line, after means after the warning line @sshaoshuai

yustaub, 2020-11-05:
I found some nan in my data, this is the reason of the runTimeWarning

sshaoshuai, 2020-11-07:
Feel free to re-open it if you still have problems.

closed 2020-11-07 by sshaoshuai