open-mmlab/OpenPCDet #343: Why not consider compensating for the timestamp interval when loading lidar data from multiple sweeps?

jialeli1, 2020-11-05:
Hi, I am trying to train a model on the nuScenes. 
And I noticed that in the function [get_lidar_with_sweeps()](https://github.com/open-mmlab/OpenPCDet/blob/0e84f82d7c45bc86ab59a646fab1da9050c05aa9/pcdet/datasets/nuscenes/nuscenes_dataset.py#L92) , the points on the adjacent frames are directly concatenated to the points of the key-frame.  However, the point cloud may have a tailing effect because the ego vehicle is moving in the time stamp interval. 
Is it appropriate to concatenate the points in multiple sweeps directly?

MartinHahner, 2020-11-05:
It would be of course better to apply motion compensation to the additional frames, but I don't know if that information is available.
However, the time delta of these additional frames is concatinated to the features, so the nuScenes models use the following features as input: `x, y, z, intensity, time_delta`.
This way the networks can maybe inherently do "motion compensation" by themselfes.

sshaoshuai, 2020-11-07:
Yes, I agree with @MartinHahner . And actually this pattern is first introduced by the official paper of NuScenes dataset. You could also refer to their paper.

closed 2020-11-07 by sshaoshuai