open-mmlab/OpenPCDet #345: About visualizing the testing data

davidwang200099, 2020-11-06:
I tried to install mayavi in my server, but failed. Must the desktop environment be installed?

brudermueller, 2020-11-06:
It should work without any mayavi deskop environment. For me it helped to execute the installation script provided in the FrustumPointnet Repo: https://github.com/charlesq34/frustum-pointnets/tree/master/mayavi 
They also provide a test script.

davidwang200099, 2020-11-06:
I mean should I install the desktop environment of the Operating System? The OS of my server is ubuntu and ubuntu desktop environment is not installed.

brudermueller, 2020-11-06:
Sorry, the question wasn't clear. I am also training on a server remotely and I am using mayavia via ssh in a jupyter notebook to visualize the result. This tutorial helped me to set it up: https://github.com/kuixu/kitti_object_vis/tree/master/jupyter

sshaoshuai, 2020-11-16:
You should either visualize with the desktop environment or just as @brudermueller said to use ssh.

closed 2020-11-17 by davidwang200099