open-mmlab/OpenPCDet #347: Error when run demo.

FanLu97, 2020-11-06:
Hello, thanks for this great toolbox.
Recently, I encountered an error when running demo.py, 
```
2020-11-06 21:50:59,378   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2020-11-06 21:50:59,413   INFO  Total number of samples: 	1
2020-11-06 21:51:01,488   INFO  ==> Loading parameters from checkpoint ../pretrain/pv_rcnn_8369.pth to CPU
2020-11-06 21:51:01,912   INFO  ==> Done (loaded 367/367)
Traceback (most recent call last):
  File "demo.py", line 109, in <module>
    main()
  File "demo.py", line 93, in main
    for idx, data_dict in enumerate(demo_dataset):
  File "demo.py", line 58, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/home/lufan/Projects/OpenPCDet/pcdet/datasets/dataset.py", line 142, in prepare_data
    if len(data_dict['gt_boxes']) == 0:
KeyError: 'gt_boxes'
```

I referred to dataset.py and found that three lines are added in the current version:
```
if len(data_dict['gt_boxes']) == 0:
     new_index = np.random.randint(self.__len__())
     return self.__getitem__(new_index)
```

I replace the code with
```
if self.training:
    if len(data_dict['gt_boxes']) == 0:
        new_index = np.random.randint(self.__len__())
        return self.__getitem__(new_index)
```
And the demo can work now.

Is this a bug in the current version or I made a mistake when run demo.py?

RanaMostafaAbdElMohsen, 2020-11-06:
Thanks a lot. I believe it is the current version and I have same issue

sshaoshuai, 2020-11-07:
Thank you for the bug report.  It is actually a bug added from my recent PR https://github.com/open-mmlab/OpenPCDet/pull/340. 
Sorry, I will fix it.

sshaoshuai, 2020-11-07:
Done.

closed 2020-11-07 by sshaoshuai