open-mmlab/OpenPCDet #348: Out of memory when training pv-rcnn model

arxidinakbar, 2020-11-07:
When training pv-rcnn model with single-gpu mode, out of memory occurs.
It shows the error message below. And it happent seceral times during training.
`GPUassert: out of memory pcdet/ops/iou3d_nms/src/iou3d_nms.cpp 103`

**my devide configuration:**  2080Ti with 11G memory
It shows that total usage is 9341M, should not be run out of memory.
**Want some clue for this issue, thank you.**

sshaoshuai, 2020-11-07:
For your GPU with 11G memory, you'd better use batch_size=2. The OOM may just happen at this line and you could not capture it by nvidia-smi.

arxidinakbar, 2020-11-09:
@sshaoshuai  Thank you so  much for early reply, solved my problem!
And there is a another question about performance, which I got higher than reported performance.
`

> Car AP@0.70, 0.70, 0.70:
> bbox AP:90.7803, 89.8999, 89.0433
> bev  AP:90.0097, 87.9282, 86.4528
> 3d   AP:88.6137, 78.6245, 77.2243
> aos  AP:90.76, 89.77, 88.82
> Car AP_R40@0.70, 0.70, 0.70:
> bbox AP:95.6261, 94.1728, 91.7683
> bev  AP:92.4184, 88.5586, 87.6479
> 3d   AP:90.5534, 81.6116, 78.6108
> aos  AP:95.59, 94.01, 91.52
> Car AP@0.70, 0.50, 0.50:
> bbox AP:90.7803, 89.8999, 89.0433
> bev  AP:90.7940, 90.1441, 89.5173
> 3d   AP:90.7940, 90.0886, 89.4014
> aos  AP:90.76, 89.77, 88.82
> Car AP_R40@0.70, 0.50, 0.50:
> bbox AP:95.6261, 94.1728, 91.7683
> bev  AP:95.6751, 94.8476, 94.2478
> 3d   AP:95.6623, 94.7450, 94.0537
> aos  AP:95.59, 94.01, 91.52
> Pedestrian AP@0.50, 0.50, 0.50:
> bbox AP:68.7347, 66.3279, 63.2579
> bev  AP:61.9979, 56.6604, 53.8126
> 3d   AP:56.5544, 52.9835, 47.7343
> aos  AP:64.66, 61.76, 58.45
> Pedestrian AP_R40@0.50, 0.50, 0.50:
> bbox AP:69.5228, 66.4894, 62.8409
> bev  AP:60.7329, 56.5680, 52.1375
> 3d   AP:55.9413, 51.1434, 46.1661
> aos  AP:64.91, 61.28, 57.51
> Pedestrian AP@0.50, 0.25, 0.25:
> bbox AP:68.7347, 66.3279, 63.2579
> bev  AP:75.3555, 73.8473, 69.7569
> 3d   AP:75.3437, 73.7890, 69.6610
> aos  AP:64.66, 61.76, 58.45
> Pedestrian AP_R40@0.50, 0.25, 0.25:
> bbox AP:69.5228, 66.4894, 62.8409
> bev  AP:76.3102, 74.7104, 70.8022
> 3d   AP:76.3025, 74.5821, 70.7215
> aos  AP:64.91, 61.28, 57.51
> Cyclist AP@0.50, 0.50, 0.50:
> bbox AP:87.5600, 77.0909, 74.3813
> bev  AP:84.0183, 70.7012, 65.4772
> 3d   AP:80.5862, 67.1589, 63.1087
> aos  AP:87.42, 76.74, 74.00
> Cyclist AP_R40@0.50, 0.50, 0.50:
> bbox AP:91.4087, 79.0226, 75.4018
> bev  AP:88.0461, 71.1604, 66.8949
> 3d   AP:82.9640, 66.7401, 62.7843
> aos  AP:91.25, 78.61, 74.96
> Cyclist AP@0.50, 0.25, 0.25:
> bbox AP:87.5600, 77.0909, 74.3813
> bev  AP:86.0192, 76.8489, 72.3789
> 3d   AP:86.0192, 76.8489, 72.3770
> aos  AP:87.42, 76.74, 74.00
> Cyclist AP_R40@0.50, 0.25, 0.25:
> bbox AP:91.4087, 79.0226, 75.4018
> bev  AP:90.2814, 77.2718, 73.5534
> 3d   AP:90.2814, 77.2718, 73.5507
> aos  AP:91.25, 78.61, 74.96`
It's wierd because I did follow the commands in README, but results are better than report.
Are you familiar with this situation? I don't know should I be happy with this outcome or I did something wrong.

sshaoshuai, 2020-11-10:
I didn't see your results. Generally speaking, you may achieve slightly better or worse results on KITTI dataset by training with the provided configs, which is a common phenomenon.

arxidinakbar, 2020-11-11:
oops, so the results I post wasn't the final results,right?
I 'm new to this, might didn't get it right, I would give it another shot. 
Thank you so much for the reply, and all the efforts.

best luck,

sshaoshuai, 2020-11-11:
It seems that your results were generated by the provided model.

closed 2020-11-11 by sshaoshuai