open-mmlab/OpenPCDet #350: not able to generate waymo_processed_data

MartinHahner, 2020-11-10:
@sshaoshuai 
Thank you for all the continued work you put into this repository!

I am facing issues when I am trying to generate `waymo_processed_data` following your instructions in [Getting Started](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md#waymo-open-dataset).

I tried it on a Tesla K80, so I thought it is maybe because the GPU is too old, 
so I tried it on a Tesla V100 as well, the two error logs are listed below.

Any idea what could be the issue?

**Tesla K80**
[K80.log](https://github.com/open-mmlab/OpenPCDet/files/5518885/K80.log) (too large to put in line)

**Tesla V100**
```
/home/ubuntu/anaconda3/envs/PCDet/lib/python3.6/runpy.py:125: RuntimeWarning: 'pcdet.datasets.waymo.waymo_dataset' found in sys.modules after import of package 'pcdet.datasets.waymo', but prior to execution of 'pcdet.datasets.waymo.waymo_dataset'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
/home/ubuntu/efs/repositories/PCDet/pcdet/datasets/waymo/waymo_dataset.py:360: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  dataset_cfg = EasyDict(yaml.load(open(args.cfg_file)))
2020-11-10 16:41:45,781   INFO  Loading Waymo dataset
2020-11-10 16:41:45,800   INFO  Total skipped info 202
2020-11-10 16:41:45,800   INFO  Total samples for Waymo dataset: 0
2020-11-10 16:41:45,800   INFO  Total sampled samples for Waymo dataset: 0
2020-11-10 16:41:46,039   INFO  Loading Waymo dataset
2020-11-10 16:41:46,074   INFO  Total skipped info 798
2020-11-10 16:41:46,074   INFO  Total samples for Waymo dataset: 0
2020-11-10 16:41:46,074   INFO  Total sampled samples for Waymo dataset: 0
2020-11-10 16:41:47.870777: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcuda.so.1
2020-11-10 16:41:47.899821: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1006] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-11-10 16:41:47.900781: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1618] Found device 0 with properties: 
name: Tesla V100-SXM2-16GB major: 7 minor: 0 memoryClockRate(GHz): 1.53
pciBusID: 0000:00:1e.0
2020-11-10 16:41:47.900824: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.0
2020-11-10 16:41:47.902699: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcublas.so.10.0
2020-11-10 16:41:47.904293: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcufft.so.10.0
2020-11-10 16:41:47.904664: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcurand.so.10.0
2020-11-10 16:41:47.906729: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusolver.so.10.0
2020-11-10 16:41:47.908282: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusparse.so.10.0
2020-11-10 16:41:47.908333: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudnn.so.7
2020-11-10 16:41:47.908423: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1006] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-11-10 16:41:47.909394: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1006] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-11-10 16:41:47.910276: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1746] Adding visible gpu devices: 0
2020-11-10 16:41:47.912516: I tensorflow/core/platform/cpu_feature_guard.cc:142] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2020-11-10 16:41:47.921158: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 2300080000 Hz
2020-11-10 16:41:47.921714: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x7fb678444140 executing computations on platform Host. Devices:
2020-11-10 16:41:47.921736: I tensorflow/compiler/xla/service/service.cc:175]   StreamExecutor device (0): Host, Default Version
2020-11-10 16:41:47.921935: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1006] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-11-10 16:41:47.922864: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1618] Found device 0 with properties: 
name: Tesla V100-SXM2-16GB major: 7 minor: 0 memoryClockRate(GHz): 1.53
pciBusID: 0000:00:1e.0
2020-11-10 16:41:47.922902: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.0
2020-11-10 16:41:47.922925: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcublas.so.10.0
2020-11-10 16:41:47.922943: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcufft.so.10.0
2020-11-10 16:41:47.922960: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcurand.so.10.0
2020-11-10 16:41:47.922977: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusolver.so.10.0
2020-11-10 16:41:47.922994: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusparse.so.10.0
2020-11-10 16:41:47.923008: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudnn.so.7
2020-11-10 16:41:47.923070: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1006] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-11-10 16:41:47.923993: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1006] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-11-10 16:41:47.924856: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1746] Adding visible gpu devices: 0
2020-11-10 16:41:50.566097: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1159] Device interconnect StreamExecutor with strength 1 edge matrix:
2020-11-10 16:41:50.566155: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1165]      0 
2020-11-10 16:41:50.566164: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1178] 0:   N 
2020-11-10 16:41:50.566443: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1006] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-11-10 16:41:50.567417: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1006] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-11-10 16:41:50.568333: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1006] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-11-10 16:41:50.569241: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1304] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 14638 MB memory) -> physical GPU (device: 0, name: Tesla V100-SXM2-16GB, pci bus id: 0000:00:1e.0, compute capability: 7.0)
2020-11-10 16:41:50.571892: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x7fb56947d210 executing computations on platform CUDA. Devices:
2020-11-10 16:41:50.571913: I tensorflow/compiler/xla/service/service.cc:175]   StreamExecutor device (0): Tesla V100-SXM2-16GB, Compute Capability 7.0
2020-11-10 16:41:52.543438: E tensorflow/stream_executor/cuda/cuda_blas.cc:428] failed to run cuBLAS routine: CUBLAS_STATUS_NOT_SUPPORTED
2020-11-10 16:41:52.543505: E tensorflow/stream_executor/cuda/cuda_blas.cc:2301] Internal: failed BLAS call, see log for details
[... cut ...]
2020-11-10 16:41:52.828395: I tensorflow/stream_executor/stream.cc:4816] [stream=0x7fb678d712a0,impl=0x7fb678d6f7d0] did not memcpy host-to-device; source: 0x7fb121f06b40
2020-11-10 16:41:52.828529: F tensorflow/core/common_runtime/gpu/gpu_util.cc:342] CPU->GPU Memcpy failed
```

The necessary imports seem to work fine:
```python
(PCDet) ubuntu /home/ubuntu/efs/repositories/PCDet \ 
→ python
Python 3.6.11 | packaged by conda-forge | (default, Aug  5 2020, 20:09:42) 
[GCC 7.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorflow as tf
>>> import waymo_open_dataset as wod
>>> print(tf.__version__)
2.0.0
```

sshaoshuai, 2020-11-10:
It seems the program could not find your cuBLAS, but I am not sure whether this problem will stop the program. 
In my side the codes could be executed well. I put my log here for your reference: 

```
2020-11-10 15:22:03,430   INFO  Loading Waymo dataset                            
2020-11-10 15:22:03,439   INFO  Total skipped info 202                                
2020-11-10 15:22:03,440   INFO  Total samples for Waymo dataset: 0             
2020-11-10 15:22:03,440   INFO  Total sampled samples for Waymo dataset: 0
2020-11-10 15:22:03,651   INFO  Loading Waymo dataset                            
2020-11-10 15:22:03,688   INFO  Total skipped info 798    
2020-11-10 15:22:03,688   INFO  Total samples for Waymo dataset: 0       
2020-11-10 15:22:03,688   INFO  Total sampled samples for Waymo dataset: 0
2020-11-10 15:25:43.274888: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcuda.so.1                                                                
2020-11-10 15:25:43.822899: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1618] Found device 0 with properties:
name: GeForce GTX 1080 Ti major: 6 minor: 1 memoryClockRate(GHz): 1.62                                                                                                                                      
pciBusID: 0000:08:00.0                           
2020-11-10 15:25:43.882832: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcudart.so.10.0'; dlerror: libcudart.so.10.0: cannot open shared object file: No such file or directory; 
2020-11-10 15:25:43.895999: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcublas.so.10.0'; dlerror: libcublas.so.10.0: cannot open shared object file:
No such file or directory;
2020-11-10 15:25:43.906020: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcufft.so.10.0'; dlerror: libcufft.so.10.0: cannot open shared object file: No such file or directory;  
2020-11-10 15:25:43.918570: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcurand.so.10.0'; dlerror: libcurand.so.10.0: cannot open shared object file:
No such file or directory; 
2020-11-10 15:25:43.928620: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcusolver.so.10.0'; dlerror: libcusolver.so.10.0: cannot open shared object file: No such file or directory;    
2020-11-10 15:25:43.939931: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcusparse.so.10.0'; dlerror: libcusparse.so.10.0: cannot open shared object file: No such file or directory; 
2020-11-10 15:25:44.901639: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudnn.so.7                                                               
2020-11-10 15:25:44.901713: W tensorflow/core/common_runtime/gpu/gpu_device.cc:1641] Cannot dlopen some GPU libraries. Please make sure the missing libraries mentioned above are installed properly if you
would like to use GPU. Follow the guide at https://www.tensorflow.org/install/gpu for how to download and setup the required libraries for your platform.                                                   
Skipping registering GPU devices...              
2020-11-10 15:25:45.281491: I tensorflow/core/platform/cpu_feature_guard.cc:142] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA                               
2020-11-10 15:25:47.671298: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 2499995000 Hz
2020-11-10 15:25:47.788556: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x7f6eaf992160 executing computations on platform Host. Devices:                                                  
2020-11-10 15:25:47.788607: I tensorflow/compiler/xla/service/service.cc:175]   StreamExecutor device (0): Host, Default Version
2020-11-10 15:25:48.025482: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1159] Device interconnect StreamExecutor with strength 1 edge matrix:                                                        
2020-11-10 15:25:48.025540: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1165]
2020-11-10 15:25:48.658699: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x7f6eaf9f56b0 executing computations on platform CUDA. Devices:                                                  
2020-11-10 15:25:48.658765: I tensorflow/compiler/xla/service/service.cc:175]   
StreamExecutor device (0): GeForce GTX 1080 Ti, Compute Capability 6.1
---------------Start to generate data infos--------------- 
---------------The waymo sample interval is 1, total sequecnes is 798-----------------
Infos are saved to (sampled_interval=1): xxx/data/waymo/waymo_processed_data/segment-10017090168044687777_6380_000_6400_000_with_camera_labels/segment-10017090168044687777_6380_000_6400_000_with_camera_labels.pkl
...
```

MartinHahner, 2020-11-10:
Thanks, I will investigate further.

MartinHahner, 2020-11-10:
I think it could be related to this issue:
https://github.com/tensorflow/tensorflow/issues/23473
Did not find a fix yet though, will look into it more the comming days...

tianweiy, 2020-11-10:
Hi, I think the generation won't need gpu. Try disable it by ```CUDA_VISIBLE_DEVICES=-1```  tensorflow doesn't work well with multithreading afaik

MartinHahner, 2020-11-11:
> Hi, I think the generation won't need gpu. Try disable it by `CUDA_VISIBLE_DEVICES=-1` tensorflow on gpu doesn't work well with multithreading afaik

Thank you @tianweiy! Your solution seems to work, now it is running...
```
(PCDet) ubuntu /home/ubuntu/efs/repositories/PCDet \ 
→ CUDA_VISIBLE_DEVICES=-1 python -m pcdet.datasets.waymo.waymo_dataset --func create_waymo_infos     --cfg_file tools/cfgs/dataset_configs/waymo_dataset.yaml
/home/ubuntu/anaconda3/envs/PCDet/lib/python3.6/runpy.py:125: RuntimeWarning: 'pcdet.datasets.waymo.waymo_dataset' found in sys.modules after import of package 'pcdet.datasets.waymo', but prior to execution of 'pcdet.datasets.waymo.waymo_dataset'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
/home/ubuntu/efs/repositories/PCDet/pcdet/datasets/waymo/waymo_dataset.py:360: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  dataset_cfg = EasyDict(yaml.load(open(args.cfg_file)))
2020-11-11 10:38:51,530   INFO  Loading Waymo dataset
2020-11-11 10:38:51,558   INFO  Total skipped info 202
2020-11-11 10:38:51,558   INFO  Total samples for Waymo dataset: 0
2020-11-11 10:38:51,558   INFO  Total sampled samples for Waymo dataset: 0
---------------Start to generate data infos---------------
2020-11-11 10:38:51,766   INFO  Loading Waymo dataset
2020-11-11 10:38:51,800   INFO  Total skipped info 798
2020-11-11 10:38:51,800   INFO  Total samples for Waymo dataset: 0
2020-11-11 10:38:51,800   INFO  Total sampled samples for Waymo dataset: 0
---------------The waymo sample interval is 1, total sequecnes is 798-----------------
2020-11-11 10:38:53.574925: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcuda.so.1
2020-11-11 10:38:53.578716: E tensorflow/stream_executor/cuda/cuda_driver.cc:318] failed call to cuInit: CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected
2020-11-11 10:38:53.578774: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:169] retrieving CUDA diagnostic information for host: ip-172-31-47-82
2020-11-11 10:38:53.578796: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:176] hostname: ip-172-31-47-82
2020-11-11 10:38:53.578867: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:200] libcuda reported version is: 418.87.1
2020-11-11 10:38:53.578916: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:204] kernel reported version is: 418.87.1
2020-11-11 10:38:53.578940: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:310] kernel version seems to match DSO: 418.87.1
2020-11-11 10:38:53.582009: I tensorflow/core/platform/cpu_feature_guard.cc:142] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2020-11-11 10:38:53.590131: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 2300085000 Hz
2020-11-11 10:38:53.590478: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x7f4b801cea50 executing computations on platform Host. Devices:
2020-11-11 10:38:53.590512: I tensorflow/compiler/xla/service/service.cc:175]   StreamExecutor device (0): Host, Default Version
```

MartinHahner, 2020-11-19:
When you run the script with `CUDA_VISIBLE_DEVICES=-1` it will crash at the ground truth database stage.

P.S.: The script ran for almost an entire week.

```
----------------Waymo info val file is saved to /home/ubuntu/efs/repositories/PCDet/data/waymo/waymo_infos_val.pkl----------------
---------------Start create groundtruth database for data augmentation---------------
2020-11-18 00:12:09,739   INFO  Loading Waymo dataset
2020-11-18 00:12:55,801   INFO  Total skipped info 2
2020-11-18 00:12:55,801   INFO  Total samples for Waymo dataset: 157965
2020-11-18 00:12:55,813   INFO  Total sampled samples for Waymo dataset: 31593
gt_database sample: 1/157965
THCudaCheck FAIL file=../aten/src/THC/THCGeneral.cpp line=51 error=38 : no CUDA-capable device is detected
Traceback (most recent call last):
  File "/home/ubuntu/anaconda3/envs/PCDet/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/ubuntu/anaconda3/envs/PCDet/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ubuntu/efs/repositories/PCDet/pcdet/datasets/waymo/waymo_dataset.py", line 368, in <module>
    processed_data_tag=dataset_cfg.PROCESSED_DATA_TAG
  File "/home/ubuntu/efs/repositories/PCDet/pcdet/datasets/waymo/waymo_dataset.py", line 344, in create_waymo_infos
    used_classes=['Vehicle', 'Pedestrian', 'Cyclist']
  File "/home/ubuntu/efs/repositories/PCDet/pcdet/datasets/waymo/waymo_dataset.py", line 278, in create_groundtruth_database
    torch.from_numpy(points[:, 0:3]).unsqueeze(dim=0).float().cuda(),
  File "/home/ubuntu/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/cuda/__init__.py", line 163, in _lazy_init
    torch._C._cuda_init()
RuntimeError: cuda runtime error (38) : no CUDA-capable device is detected at ../aten/src/THC/THCGeneral.cpp:51
```

sshaoshuai, 2020-11-19:
The create_gt_database needs the GPU to calculate the inside points, but it doesn't need the tensorflow enviroment. 

So you could comment the following codes and just run the create_gt_database with GPU (such as, set CUDA_VISIBLE_DIEVCES=0): 
https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/waymo/waymo_dataset.py#L318-L338

MartinHahner, 2020-11-19:
[image]

Can you confirm that the highlighted numbers above are correct?

Because in `/home/ec2-user/efs/repositories/PCDet/data/waymo/waymo_processed_data`
I only have 999 `.pkl` files. Shouldn't it be 1000?

sshaoshuai, 2020-11-19:
There are two problems: 
1. It should be 1000 pkl. The `total skipped info info 2` means there are two infos that are lost. 
2. `40077` is the number of samples of the validation set, but actually it is the `version 1.0` of WOD, currently the `version 1.2` of val of WOD should be `39987`. There are little differences between the training of version 1.0 and version 1.2, but you'd better use the latest validation set with `39987` samples. 

BTW, I use ~37 hours to generate all the infos, except the create_gt_database. I am not sure whether the  speed is related to the version of your tensorflow and the waymo-open-dataset.

DeepakVellampalli, 2020-11-21:
@sshaoshuai @MartinHahner 
Iam able to generate database with ur code. Attaching the logs of database generation
[database_log.txt](https://github.com/open-mmlab/OpenPCDet/files/5577803/database_log.txt)

sshaoshuai, 2020-11-22:
@DeepakVellampalli , it seems that the training data is fine, but there are 4 skipped infos in your validation set.

chowkamlee81, 2020-11-23:
@sshaoshuai 
Thanks for pointing out the errors in my logs.

closed 2020-11-23 by sshaoshuai

d33dler, 2022-05-11:
> There are two problems:
>
> 1. It should be 1000 pkl. The `total skipped info info 2` means there are two infos that are lost.
> 2. `40077` is the number of samples of the validation set, but actually it [...]

@sshaoshuai does this data info generator use multithreading? want to run it on a cluster, but 37 hours seems like a waste