open-mmlab/OpenPCDet #355: gt_sampling seems consume large memory

Nightmare-n, 2020-11-18:
2020-11-18 15:53:28,693   INFO  **********************Start training waymo_models/pointpillar(default)**********************
Traceback (most recent call last):                                                                                                                                                                                 
  File "train.py", line 198, in <module>
    main()
  File "train.py", line 170, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/envtensor/OpenPCDet/tools/train_utils/train_utils.py", line 93, in train_model
    dataloader_iter=dataloader_iter
  File "/envtensor/OpenPCDet/tools/train_utils/train_utils.py", line 12, in train_one_epoch
    dataloader_iter = iter(train_loader)
  File "/opt/conda/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 291, in __iter__
    return _MultiProcessingDataLoaderIter(self)
  File "/opt/conda/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 737, in __init__
    w.start()
  File "/opt/conda/lib/python3.7/multiprocessing/process.py", line 112, in start
    self._popen = self._Popen(self)
  File "/opt/conda/lib/python3.7/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/opt/conda/lib/python3.7/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
  File "/opt/conda/lib/python3.7/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/opt/conda/lib/python3.7/multiprocessing/popen_fork.py", line 20, in __init__
    self._launch(process_obj)
  File "/opt/conda/lib/python3.7/multiprocessing/popen_spawn_posix.py", line 59, in _launch
    cmd, self._fds)
  File "/opt/conda/lib/python3.7/multiprocessing/util.py", line 432, in spawnv_passfds
    False, False, None)
OSError: [Errno 12] Cannot allocate memory

Does augmentation method 'gt_sampling' consume large memory in waymo dataset? if I remove it, it runs without problem.

Nightmare-n, 2020-11-18:
it's the initial memory state without training openpcdet

free -m

              total        used        free      shared  buff/cache   available
Mem:         128819       24280       94415        1439       10123      100367
Swap:         65535       65535           0

nvidia-smi

|   0  TITAN V             Off  | 00000000:04:00.0  On |                  N/A |
| 31%   46C    P8    27W / 250W |     19MiB / 12036MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  TITAN V             Off  | 00000000:05:00.0  On |                  N/A |
| 36%   54C    P2    46W / 250W |   1731MiB / 12036MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   2  TITAN V             Off  | 00000000:08:00.0  On |                  N/A |
| 33%   45C    P8    26W / 250W |      8MiB / 12036MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   3  TITAN V             Off  | 00000000:09:00.0  On |                  N/A |
| 34%   47C    P8    29W / 250W |      8MiB / 12036MiB |      0%      Default |

tianweiy, 2020-11-19:
it will increase the cpu memory for the dataloader. I guess you can decrease the num_worker.

Nightmare-n, 2020-11-20:
> it will increase the cpu memory for the dataloader. I guess you can decrease the num_worker.
> 
> ```
>     parser.add_argument('--workers', type=int, default=8, help='number of workers for dataloader')
> ```

thanks, it works.

closed 2020-11-20 by sshaoshuai