open-mmlab/OpenPCDet #356: Data-preprocessing like SECOND

tojimahammatov, 2020-11-19:
Hi, it was great work.

I saw that in nuscenes_util.py you changed pre-processing and evaluation methods from det3d and second.

But I need pre-processing like exactly SECOND, what is the fast and easy way to make it using your code? Maybe you have some configurations to change it ?! I just need images dict including their points and masks.

sshaoshuai, 2020-11-22:
You could change the parts of data preparation to SECOND, and it will not create problems for the rest training and evaluation. 
But I didn't have a configurations to generate exactly the same data with SECOND, and you need to check it by yourself.

tojimahammatov, 2020-11-23:
I understand, thank you very much.

closed 2020-11-23 by tojimahammatov