open-mmlab/OpenPCDet #358: Visulisation Issue via using mayavi

neisakutou, 2020-11-19:
Really appreciate ur work on this great platform.

Well, after I successfully finished the installation and went for the quick demo locally, I met some troubles.

Basically, I did install mayavi & qt by instructions.

The thing is that: 
- I could run the demo.py without visualisation and easily get the prediction information.
- But It returns **Segmentation fault (core dumped)** with plotting results.
So, which I figured out is that should be issues of plotting.

Here is the environment:
cuda: 9.0.176
pytorch: 1.1.0
python: 3.6.6
spconv: v1.0
mayavi: 4.7.2 (via pip)
PyQt5: 5.15.1 (via pip)

Thanks a lot.

sshaoshuai, 2020-11-20:
It should be the problem of mayavi.  Maybe you could first try to use some demos of mayavi to check whether it could visualize points sucessfully.

neisakutou, 2020-11-20:
> It should be the problem of mayavi.  Maybe you could first try to use some demos of mayavi to check whether it could visualize points sucessfully. 

Cheers.
Thanks for ur reply and will update anything new.

sshaoshuai, 2020-11-22:
There may exist some other tools for visualization of point clouds, such as PCL, but I never used them.

neisakutou, 2020-11-24:
> There may exist some other tools for visualization of point clouds, such as PCL, but I never used them. 

Updates!
Considering of the crash of using mayavi, alternatively, I switched it to Open3d.
And it works.
Thank u so much for ur reply and response.

K-Young96, 2021-01-19:
sudo pip3 install pyqt5==5.10.1 -i https://pypi.doubanio.com/simple

K-Young96, 2021-01-19:
> sudo pip3 install pyqt5==5.10.1 -i https://pypi.doubanio.com/simple

must use '5.10.1'

tsbhun98, 2021-02-11:
I had the same problem and it was caused by PyQt. After I removed it and installed PySide instead everything worked for me.

chenxyyy, 2021-07-01:
you can use ```open3d``` instead of ```mayavi``` to display the inference result. It is much easier than ```mayavi``` .

yinggo, 2021-10-05:
I changed the pyqt5 version by  `pip3 install pyqt5==5.10.1 -i https://pypi.doubanio.com/simple`
and in `demo.py`, I moved `import mayavi.mlab as mlab` before 'V.draw_scenes', it works.

closed 2022-03-08 by github-actions[bot]