open-mmlab/OpenPCDet #359: More datasets

dkoguciuk, 2020-11-19:
Hi OpenPCDDet community,

In the readme there is a statement:
```We are actively updating this repo currently, and more datasets and models will be supported soon. Contributions are also welcomed.```

Do you have any roadmaps / current list of work-in-progress features? I am interested in Pandaset support, is anyone working on it? Any tips, where to start?

Best,
D

jihanyang, 2020-11-20:
We will add Lyft to our codebase soon. Pandaset is also under our consideration. Thanks for your advice, we wiil consider to add a work-in-progress features.

dkoguciuk, 2020-11-20:
Ok, great. Do you have any tips on how to work on adding a new dataset? Maybe I could give it a try.

Best,
D

sshaoshuai, 2020-11-22:
You could refer to the recently added Waymo Dataset to see how to add new datasets. 
Simply speaking, you just need to:
1. organize your data with training / validation.
2. transform the point clouds to our unified normative coordinate at the __getitem__. 
3. call the prepare_data function. 
4. receive the predicted results. 
3. write your evaluation codes.

closed 2022-03-08 by github-actions[bot]