open-mmlab/OpenPCDet #360: python demo.py pv_rcnn.yaml error occurs，however pointpillar.yaml、pointrcnn.yaml ,OK ,why #3

wangwei-gpu, 2020-11-20:
python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --ckpt pth/pv_rcnn_8369.pth --data_path ../testing/velodyne/000006.bin 。The following error occurs：
INFO Visualized sample index: 1
Traceback (most recent call last):
File "demo.py", line 103, in
main()
File "demo.py", line 91, in main
pred_dicts, _ = model.forward(data_dict)
File "/home/dfmc/OpenPCDet/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
batch_dict = cur_module(batch_dict)
File "/home/dfmc/anaconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 541, in call
result = self.forward(*input, **kwargs)
File "/home/dfmc/OpenPCDet/pcdet/models/backbones_3d/spconv_backbone.py", line 139, in forward
x = self.conv_input(input_sp_tensor)
File "/home/dfmc/anaconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 541, in call
result = self.forward(*input, **kwargs)
File "/home/dfmc/.local/lib/python3.6/site-packages/spconv/modules.py", line 134, in forward
input = module(input)
File "/home/dfmc/anaconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 541, in call
result = self.forward(*input, **kwargs)
File "/home/dfmc/.local/lib/python3.6/site-packages/spconv/conv.py", line 181, in forward
use_hash=self.use_hash)
File "/home/dfmc/.local/lib/python3.6/site-packages/spconv/ops.py", line 95, in get_indice_pairs
int(use_hash))
ValueError: /home/dfmc/spconv-master/src/spconv/spconv_ops.cc 87
unknown device type，What is the problem？ thank you very much

sshaoshuai, 2020-11-20:
Have you installed correct `spconv`?  Neither pointpillar nor pointrcnn use spconv. 
So the reason may be that you do not have a proper installed spconv.

cowarder, 2020-11-29:
Hi, did you solve this problem? I meet the same error as you, 'import spconv' works fine.

wangwei-gpu, 2020-12-07:
OK ，thank you

closed 2020-12-07 by wangwei-gpu

Shangyi-Li, 2023-03-24:
Well, I tried to rebuild and reinstall spconv lib, but the "unknown device type" problem remains.
I wish anybody could help me.