open-mmlab/OpenPCDet #365: Error when training PV-RCNN on multiple GPUs

raviteja-kvns, 2020-11-22:
I am successfully able to train PV-RCNN on single GPU. I am able to train other networks say pointrcnn on multiple gpus. However when I try to train PV-RCNN on multiple GPUs, I get the following error:

2020-11-16 18:07:54,162   INFO  **********************Start training kitti_models/pv_rcnn_ours_10(default)**********************

epochs:   0%|          | 0/80 [00:00<?, ?it/s]
epochs:   0%|          | 0/80 [00:00<?, ?it/s]
epochs:   0%|          | 0/80 [00:00<?, ?it/s]
epochs:   0%|          | 0/80 [00:00<?, ?it/s]

train:   0%|          | 0/988 [00:00<?, ?it/s][A
epochs:   0%|          | 0/80 [00:11<?, ?it/s]
Traceback (most recent call last):
  File "train.py", line 205, in <module>
    main()
  File "train.py", line 177, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/team1/codes/3dObjDet/OpenPCDet_ravi/tools/train_utils/train_utils.py", line 100, in train_model
    dataloader_iter=dataloader_iter
  File "/team1/codes/3dObjDet/OpenPCDet_ravi/tools/train_utils/train_utils.py", line 38, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/team1/codes/3dObjDet/OpenPCDet_ravi/pcdet/models/__init__.py", line 30, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/parallel/distributed.py", line 376, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/team1/codes/3dObjDet/OpenPCDet_ravi/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/team1/codes/3dObjDet/OpenPCDet_ravi/pcdet/models/backbones_3d/spconv_backbone.py", line 139, in forward
    x = self.conv_input(input_sp_tensor)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/spconv/modules.py", line 127, in forward
    input.features = module(input.features)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/batchnorm.py", line 83, in forward
    exponential_average_factor, self.eps)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/functional.py", line 1697, in batch_norm
    training, momentum, eps, torch.backends.cudnn.enabled
RuntimeError: cuDNN error: CUDNN_STATUS_EXECUTION_FAILED

                                              
Traceback (most recent call last):
  File "train.py", line 205, in <module>
    main()
  File "train.py", line 177, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/team1/codes/3dObjDet/OpenPCDet_ravi/tools/train_utils/train_utils.py", line 100, in train_model
    dataloader_iter=dataloader_iter
  File "/team1/codes/3dObjDet/OpenPCDet_ravi/tools/train_utils/train_utils.py", line 38, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/team1/codes/3dObjDet/OpenPCDet_ravi/pcdet/models/__init__.py", line 30, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/parallel/distributed.py", line 376, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/team1/codes/3dObjDet/OpenPCDet_ravi/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/team1/codes/3dObjDet/OpenPCDet_ravi/pcdet/models/backbones_3d/spconv_backbone.py", line 139, in forward
    x = self.conv_input(input_sp_tensor)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/spconv/modules.py", line 127, in forward
    input.features = module(input.features)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/batchnorm.py", line 83, in forward
    exponential_average_factor, self.eps)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/functional.py", line 1697, in batch_norm
    training, momentum, eps, torch.backends.cudnn.enabled
RuntimeError: cuDNN error: CUDNN_STATUS_EXECUTION_FAILED

                                              [ATraceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.6/dist-packages/torch/distributed/launch.py", line 235, in <module>
    main()
  File "/usr/local/lib/python3.6/dist-packages/torch/distributed/launch.py", line 231, in main
    cmd=process.args)
subprocess.CalledProcessError: Command '['/usr/local/bin/python', '-u', 'train.py', '--local_rank=0', '--launcher', 'pytorch', '--cfg_file', 'cfgs/kitti_models/pv_rcnn_ours_10.yaml']' returned non-zero exit status 1.

NLCharles, 2020-11-24:
Check if torch.cuda.is_available() is True. Sometimes it switches to False due to system changes or device occupancy. Try to restart your shell or system?

LZDSJTU, 2020-11-27:
@raviteja-kvns Did you use the latest version? I also met such problem that I can train with one GPU but fail with multi GPU. However, I did not get the error information as I run the code in the server. I do not meet the problem in the previous version.

raviteja-kvns, 2020-12-02:
Thank you @NLCharles and @LZDSJTU for your suggestions. Turns out the issue is with batch size 2 i.e. GPU memory is not sufficient. When I change batch size to 1, multi-gpu training works fine.

closed 2020-12-02 by raviteja-kvns

ys920826, 2021-04-03:
Hello, I have encountered a similar error when training with multiple 2080ti GPUs. What is your GPU model you use, how much GPU training solves this problem, thank you