open-mmlab/OpenPCDet #369: importing Onnx models to Pytorch?

ramchandraKarandikar, 2020-11-23:
Hi I have a prebuilt model in onnx format and I wanted to convert / import it to pytorch. Does OpenPCDet support this operation?

sshaoshuai, 2020-11-24:
No, we haven't supported this feature currently.
Duplicated with https://github.com/open-mmlab/OpenPCDet/issues/228.

ramchandraKarandikar, 2020-11-24:
Okay, This can be a separate issue than the one referenced above: #228 Since this issue talks about converting onnx models back to pytorch. The referenced issue talks about exporting to pytorch (if I am not mistaken)

sshaoshuai, 2020-11-25:
I see, welcome to submit a PR if you would like to add this feature. 
But I think only few 3D detection methods could be transformed between onnx and pytorch since there are many customized operations.

closed 2022-03-08 by github-actions[bot]