open-mmlab/OpenPCDet #370: Low accuracy when training on waymo dataset

LZDSJTU, 2020-11-24:
Hi, I add the recent update on the waymo dataset to the previous version of OpenPCDet (maybe June2020 version), and train second/PartA2 on the waymo dataset. I do not modify the preprocess script and model cfg for waymo.

However, the accuracy is much lower. For second, the final accuracy is as follows:
742 2020-11-24 08:00:32,763   INFO  ****************Evaluation done.*****************
743 2020-11-24 09:47:11,516   INFO  *************** EPOCH 0 EVALUATION *****************
744 2020-11-24 09:48:05,627   INFO  *************** Performance of EPOCH 0 *****************
745 2020-11-24 09:48:05,627   INFO  Generate label finished(sec_per_example: 0.0068 second).
746 2020-11-24 09:48:05,628   INFO  recall_roi_0.3: 0.000000 (0 / 296561)
747 2020-11-24 09:48:05,628   INFO  recall_rcnn_0.3: 0.689605 (204510 / 296561)
748 2020-11-24 09:48:05,628   INFO  recall_roi_0.5: 0.000000 (0 / 296561)
749 2020-11-24 09:48:05,628   INFO  recall_rcnn_0.5: 0.574320 (170321 / 296561)
750 2020-11-24 09:48:05,628   INFO  recall_roi_0.7: 0.000000 (0 / 296561)
751 2020-11-24 09:48:05,628   INFO  recall_rcnn_0.7: 0.316451 (93847 / 296561)
752 2020-11-24 09:48:05,637   INFO  Average predicted number of objects(7998 samples): 100.557
753 2020-11-24 09:51:25,703   INFO
754 OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.3911
755 OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.3855
756 OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.3620
757 OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.3568
758 OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.3631
759 OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.2795
760 OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.3308
761 OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.2545
762 OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000
763 OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000
764 OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000
765 OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000
766 OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.2761
767 OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.2636
768 OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.2757
769 OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.2632

I'd like to ask what may cause this problem.

Thank you very much.

LZDSJTU, 2020-11-24:
I trained 80 epoches, as the previous version does not load the "NUM_EPOCHS" parameter in the cfg

sshaoshuai, 2020-11-24:
What's your model for evaluation? It seems you used the epoch 0 for evaulation.

LZDSJTU, 2020-11-24:
> What's your model for evaluation? It seems you used the epoch 0 for evaulation.

I used the last epoch to evaluate, which is the online evaluation during the training process. This is just a typo error.

LZDSJTU, 2020-11-24:
And I do not use "merge_all_iters_to_one_epoch".

sshaoshuai, 2020-11-24:
Let me check whether there is a bug in the dataloader.

LZDSJTU, 2020-11-24:
> Let me check whether there is a bug in the dataloader.

Thank you.

sshaoshuai, 2020-11-24:
Currently I didn't find any obvious bug, mayby you could try to train with the latest codes. 
I will also try to train a model.

LZDSJTU, 2020-11-24:
> Currently I didn't find any obvious bug, mayby you could try to train with the latest codes.
> I will also try to train a model.

Okay

sshaoshuai, 2020-11-25:
I got much better results than your reported performance, hence it should not be a problem of the dataloader. 
I will try more models since the released dataloader is a little bit different with me since my dataloader is an old version.

LZDSJTU, 2020-11-25:
I just donwload the updated version and try to train it. However, out-of-memory issue happened. I do not know if the updated version need much more memory for the waymo dataset.

sshaoshuai, 2020-11-25:
No, I could train SECOND with the given config on GPU 1080Ti with 11G memory.

LZDSJTU, 2020-11-25:
I use the latest version. My training will always be stuck here：
2020-11-25 15:59:13,004   INFO  **********************Start training data1/OpenPCDet-master20201117/tools/cfgs/waymo_models/second(test_in_shell)**********************
epochs:   0%|                                                                                                                                                     | 0/30 [00:00<?, ?it/s]

LZDSJTU, 2020-11-25:
It seems that the problem is caused by the multi-gpu training. When I use the single gpu, the training is OK. This problem does not happen when I use the old version. I will check it.

LZDSJTU, 2020-11-25:
https://github.com/open-mmlab/OpenPCDet/blob/594962844729353842207b9e3c50f8b2434484f9/tools/train_utils/train_utils.py#L76

When using two gpus for training, the code is stack here. When using the old version in the same device, I think that maybe the dataloader cause the problem. Is the updated dataloader much different from the old one?

LZDSJTU, 2020-11-27:
When I set the num_worker=0, multi-gpu training is ok. Such problem does not exsit in the previous version. I will look for the cause of this situation.

sshaoshuai, 2020-11-27:
@LZDSJTU , I do find some problems in the evaluation codes, please see https://github.com/open-mmlab/OpenPCDet/pull/384. 
But the data and training codes are fine. 
You could follow the instructions to re-create the validation infos and re-evaluate your models.

LZDSJTU, 2020-11-27:
OK, I will modify the evaluation code. I think that the latest version has some problem in dataloader. I run different versions in the same environment with multi-gpus, the previous version is OK but the current version is deadlock. I haven't found the reason.

sshaoshuai, 2020-11-27:
Have you tried to run with other datasets like KITTI?

LZDSJTU, 2020-11-27:
Yep, the similar problem still exsits for KITTI. The difference is that using KITTI may not meet deadlock. It just gets stack for a long time and start running. However, if using the previous version, it is very fast to start running.

MartinHahner, 2020-11-27:
> I do find some problems in the evaluation codes, please see #384.
> But the data and training codes are fine.
> You could follow the instructions to re-create the validation infos and re-evaluate your models.

Saw this message just in time minutes after my script started to create them. 
I had to re-download WOD (I used to work with v1.0 instead of v1.2). 
I will try to reproduce your results on WOD next week.

LZDSJTU, 2020-11-30:
> @LZDSJTU , I do find some problems in the evaluation codes, please see #384.
> But the data and training codes are fine.
> You could follow the instructions to re-create the validation infos and re-evaluate your models.

Hi, I re-preprocess the waymo val set and evaluate the accuracy. Actually, the accuracy increases but is still much lower than yours. I trained it with the old pcdet with the waymo cfg.

I have not trained on the latest version as the multi-gpu training has some problem in my environment before. I found that If I insert the following sentence in eval_utils.py, the multi-gpu training will be okay:

`from pcdet.datasets.kitti.kitti_object_eval_python.eval import print_str`

I cannot understand the reason now as the "print_str" is not used in eval_utils.py and my code is not stuck here. It got stuck at the beginning of training.

LZDSJTU, 2020-12-01:
@sshaoshuai @MartinHahner 
I train the second model using the latest version, and I have re-created the val info for the waymo dataset.

I follow the instruction to train the model for 30 epochs. The final accuracy is still much lower:
5 2020-12-01 03:47:44,740   INFO  *************** EPOCH 30 EVALUATION *****************
366 2020-12-01 03:48:32,127   INFO  *************** Performance of EPOCH 30 *****************
367 2020-12-01 03:48:32,128   INFO  Generate label finished(sec_per_example: 0.0059 second).
368 2020-12-01 03:48:32,128   INFO  recall_roi_0.3: 0.000000
369 2020-12-01 03:48:32,128   INFO  recall_rcnn_0.3: 0.668910
370 2020-12-01 03:48:32,128   INFO  recall_roi_0.5: 0.000000
371 2020-12-01 03:48:32,128   INFO  recall_rcnn_0.5: 0.552218
372 2020-12-01 03:48:32,128   INFO  recall_roi_0.7: 0.000000
373 2020-12-01 03:48:32,128   INFO  recall_rcnn_0.7: 0.297340
374 2020-12-01 03:48:32,133   INFO  Average predicted number of objects(7998 samples): 101.049
375 2020-12-01 03:51:04,877   INFO
376 OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.4284
377 OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.4213
378 OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.3733
379 OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.3670
380 OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.4020
381 OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.3015
382 OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.3456
383 OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.2589
384 OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000
385 OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000
386 OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000
387 OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000
388 OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.2885
389 OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.2714
390 OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.2791
391 OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.2626

The result is very similar with that using the previous PCDet version which adds the waymo cfg (after re-creating the val info).

LZDSJTU, 2020-12-01:
Currently I only re-create the val info for the waymo dataset. I do not know whether the train info needs to be re-created as the gt_sample needs to the parameter 'num_points_in_gt' to filter the database.

But my accuracy is very different from the your provided, so I am not sure whether the gap is caused by this reason.

LZDSJTU, 2020-12-01:
This is my training log for SECOND model.
-----------------------------------------------------------------------------------------------
2020-11-30 19:51:53,592   INFO  **********************Start logging**********************
2020-11-30 19:51:53,592   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-11-30 19:51:53,592   INFO  total_batch_size: 32
2020-11-30 19:51:53,592   INFO  cfg_file         /data1/OpenPCDet-master_new/tools/cfgs/waymo_models/second.yaml
2020-11-30 19:51:53,592   INFO  batch_size       4
2020-11-30 19:51:53,592   INFO  epochs           30
2020-11-30 19:51:53,592   INFO  workers          8
2020-11-30 19:51:53,592   INFO  extra_tag        20201130_second
2020-11-30 19:51:53,592   INFO  ckpt             None
2020-11-30 19:51:53,592   INFO  pretrained_model None
2020-11-30 19:51:53,592   INFO  launcher         pytorch
2020-11-30 19:51:53,592   INFO  tcp_port         18865
2020-11-30 19:51:53,592   INFO  sync_bn          False
2020-11-30 19:51:53,592   INFO  fix_random_seed  False
2020-11-30 19:51:53,593   INFO  ckpt_save_interval 5
2020-11-30 19:51:53,593   INFO  local_rank       0
2020-11-30 19:51:53,593   INFO  max_ckpt_save_num 30
2020-11-30 19:51:53,593   INFO  merge_all_iters_to_one_epoch False
2020-11-30 19:51:53,593   INFO  set_cfgs         None
2020-11-30 19:51:53,593   INFO  max_waiting_mins 0
2020-11-30 19:51:53,593   INFO  start_epoch      0
2020-11-30 19:51:53,593   INFO  save_to_file     False
2020-11-30 19:51:53,593   INFO  cfg.ROOT_DIR: /data1/OpenPCDet-master_new
2020-11-30 19:51:53,593   INFO  cfg.LOCAL_RANK: 0
2020-11-30 19:51:53,593   INFO  cfg.CLASS_NAMES: ['Vehicle', 'Pedestrian', 'Cyclist']
2020-11-30 19:51:53,593   INFO  
cfg.DATA_CONFIG = edict()
2020-11-30 19:51:53,593   INFO  cfg.DATA_CONFIG.DATASET: WaymoDataset
2020-11-30 19:51:53,593   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/waymo
2020-11-30 19:51:53,593   INFO  cfg.DATA_CONFIG.PROCESSED_DATA_TAG: waymo_processed_data
2020-11-30 19:51:53,593   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [-75.2, -75.2, -2, 75.2, 75.2, 4]
2020-11-30 19:51:53,593   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2020-11-30 19:51:53,593   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2020-11-30 19:51:53,593   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2020-11-30 19:51:53,593   INFO  
cfg.DATA_CONFIG.SAMPLED_INTERVAL = edict()
2020-11-30 19:51:53,593   INFO  cfg.DATA_CONFIG.SAMPLED_INTERVAL.train: 5
2020-11-30 19:51:53,593   INFO  cfg.DATA_CONFIG.SAMPLED_INTERVAL.test: 5
2020-11-30 19:51:53,593   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2020-11-30 19:51:53,593   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2020-11-30 19:51:53,594   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['pcdet_waymo_dbinfos_train_sampled_10.pkl'], 'PREPARE': {'filter_by_min_points': ['Vehicle:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Vehicle:15', 'Pedestrian:10', 'Cyclist:10'], 'NUM_POINT_FEATURES': 5, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': True}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x', 'y']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2020-11-30 19:51:53,594   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2020-11-30 19:51:53,594   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2020-11-30 19:51:53,594   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity', 'elongation']
2020-11-30 19:51:53,594   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity', 'elongation']
2020-11-30 19:51:53,594   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': True}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.1, 0.1, 0.15], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 80000, 'test': 90000}}]
2020-11-30 19:51:53,594   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/waymo_dataset.yaml
2020-11-30 19:51:53,594   INFO  
cfg.MODEL = edict()
2020-11-30 19:51:53,594   INFO  cfg.MODEL.NAME: SECONDNet
2020-11-30 19:51:53,594   INFO  
cfg.MODEL.VFE = edict()
2020-11-30 19:51:53,594   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2020-11-30 19:51:53,595   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2020-11-30 19:51:53,595   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2020-11-30 19:51:53,595   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2020-11-30 19:51:53,595   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2020-11-30 19:51:53,595   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2020-11-30 19:51:53,595   INFO  
cfg.MODEL.BACKBONE_2D = edict()
2020-11-30 19:51:53,595   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
2020-11-30 19:51:53,595   INFO  cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [5, 5]
2020-11-30 19:51:53,595   INFO  cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [1, 2]
[... 301 lines cut ...]
2020-12-01 03:48:32,128   INFO  recall_roi_0.7: 0.000000
2020-12-01 03:48:32,128   INFO  recall_rcnn_0.7: 0.297340
2020-12-01 03:48:32,133   INFO  Average predicted number of objects(7998 samples): 101.049
2020-12-01 03:51:04,877   INFO  
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.4284 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.4213 
[... cut ...]
2020-12-01 03:51:04,878   INFO  Result is save to /data1/OpenPCDet-master_new/output/data1/OpenPCDet-master_new/tools/cfgs/waymo_models/second/20201130_second/eval/eval_with_train/epoch_30/val
2020-12-01 03:51:04,878   INFO  ****************Evaluation done.*****************
2020-12-01 03:51:04,910   INFO  Epoch 30 has been evaluated
2020-12-01 03:51:34,941   INFO  **********************End evaluation data1/OpenPCDet-master_new/tools/cfgs/waymo_models/second(20201130_second)**********************

sshaoshuai, 2020-12-01:
It seems that your data has some problems, the number of gt_database filter should be similar with the following: 
```
 129 2020-11-24 15:55:00,019   INFO  Database filter by min points Vehicle: 477761 => 392931
 130 2020-11-24 15:55:00,064   INFO  Database filter by min points Pedestrian: 222675 => 175422
 131 2020-11-24 15:55:00,069   INFO  Database filter by min points Cyclist: 5344 => 4638
 132 2020-11-24 15:55:00,220   INFO  Database filter by difficulty Vehicle: 392931 => 392931
 133 2020-11-24 15:55:00,286   INFO  Database filter by difficulty Pedestrian: 175422 => 175422
 134 2020-11-24 15:55:00,288   INFO  Database filter by difficulty Cyclist: 4638 => 4638
 135 2020-11-24 15:55:00,670   INFO  Loading Waymo dataset
 136 2020-11-24 15:55:17,695   INFO  Total skipped info 0
 137 2020-11-24 15:55:17,696   INFO  Total samples for Waymo dataset: 158081
 138 2020-11-24 15:55:17,702   INFO  Total sampled samples for Waymo dataset: 31617

```

It may be caused by your enviroment of tensorflow and waymo_open_dataset. 
Please try to re-create a new enviroment with the following command (You could change the version if you find there are some problems of your current version): 
```
# tf 2.1.0.
pip install waymo-open-dataset-tf-2-1-0==1.2.0
# tf 2.0.0
# pip install waymo-open-dataset-tf-2-0-0==1.2.0
# tf 1.15.0
# pip install waymo-open-dataset-tf-1-15-0==1.2.0
```

LZDSJTU, 2020-12-01:
> It seems that your data has some problems, the number of gt_database filter should be similar with the following:
>
> ```
>  129 2020-11-24 15:55:00,019   INFO  Database filter by min points Vehicle: [...]

Yes, using tf 2.0.0 will cause wrong preprocessed results. Using tf2.1.1 seems to be right. I will re-process the waymo dataset and train the model.

MartinHahner, 2020-12-01:
Am I getting you right, that the instructions listed [here](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md#waymo-open-dataset ) are not correct and one **has to** use 
`pip install waymo-open-dataset-tf-2-1-0==1.2.0` 
instead of 
`pip3 install waymo-open-dataset-tf-2-0-0==1.2.0`?

sshaoshuai, 2020-12-01:
Not exactly. 
Actually I tried to create a new environment and use the following for generating data: 
```
pip3 install waymo-open-dataset-tf-2-0-0==1.2.0
```
and I could get the results correctly. 

You could visualize some generated xxxx.npy files to see whether the point clouds are reasonable. If so, your environment should be fine. 

I give a reference here, you could find the following file: 
```
waymo_processed_data/segment-10017090168044687777_6380_000_6400_000_with_camera_labels/0000.npy
```
and load it with the following line: 
```
A = np.load('0000.npy')
print(A.shape)
print(A.sum())
```
it should be `(171606, 6)` and `609530.0`. 

If these numbers are matched, your data should be fine.

MartinHahner, 2020-12-01:
@sshaoshuai 
Thx! My numbers match with yours and I used 
`pip3 install waymo-open-dataset-tf-2-0-0==1.2.0` from the [Getting Started](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md#waymo-open-dataset) page.

My `waymo_processed_data` is still getting copied to AWS since I had issues generating it there.
Once all necessary data is there, I will try to validate your results as mentioned before.

JunrQ, 2020-12-09:
I followed the readme and tried to reproduce the results. I trained a `Part-A^2-Anchor` and tested it using the `tools/test.py`
Before the bugfix:
V 1 : AP / APH | 2 : AP / APH	P 1 : AP / APH | 2 : AP / APH	C 1 : AP / APH | 2 : AP / APH
0.6197 / 0.6150 | 0.5721 / 0.5677	0.5428 / 0.4706 | 0.4981 / 0.4312	0.5856 / 0.5751 | 0.5846 / 0.5741

After the bugfix:
0.7198 / 0.7144 | 0.6447 / 0.6396	0.6348 / 0.5508 | 0.5454 / 0.4722 0.6562 / 0.6444 | 0.6314 / 0.6201

which is very close to the reported result
71.82/71.29 | 64.33/63.82 | 63.15/54.96 | 54.24/47.11 | 65.23/63.92 | 62.61/61.35

jianrenw, 2020-12-18:
> Not exactly.
> Actually I tried to create a new environment and use the following for generating data:
> 
> ```
> pip3 install waymo-open-dataset-tf-2-0-0==1.2.0
> ```
> 
> and I could get the results correctly.
> 
> You could visualize some generated xxxx.npy files to see whether the point clouds are reasonable. If so, your environment should be fine.
> 
> I give a reference here, you could find the following file:
> 
> ```
> waymo_processed_data/segment-10017090168044687777_6380_000_6400_000_with_camera_labels/0000.npy
> ```
> 
> and load it with the following line:
> 
> ```
> A = np.load('0000.npy')
> print(A.shape)
> print(A.sum())
> ```
> 
> The results should be `(171606, 6)` and `609530.0`.
> 
> If these numbers are matched, your data should be fine.

I tried all versions of waymo-open-dataset, the A.sum()s are all different. But none of them equal to yours. Do you have any idea what might be the problem?

jialeli1, 2021-01-16:
> I followed the readme and tried to reproduce the results. I trained a `Part-A^2-Anchor` and tested it using the `tools/test.py`
> Before the bugfix:
> V 1 : AP / APH | 2 : AP / APH P 1 : AP / APH | 2 : [...]

Hi, do you mean this bugfix #384 ?
if I need to re-create the val info for the waymo dataset, whether I only need to keep the line333-349 in the function _create_waymo_infos()_ of _waymo_dataset.py_? Because train info is too time-consuming, I want to be sure I will not modify it.

https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/waymo/waymo_dataset.py#L309

Thanks.

MartinHahner, 2021-01-20:
I want to confirm (most of) the numbers given in [README.md](https://github.com/open-mmlab/OpenPCDet/blob/master/README.md).

Numbers provided by OpenPCDet on the full evaluation set:
|                                             | Vec_L1 | Vec_L2 | Ped_L1 | Ped_L2 | Cyc_L1 | Cyc_L2 |  
|---------------------------------------------|----------:|:-------:|:-------:|:-------:|:-------:|:-------:|
| [SECOND](tools/cfgs/waymo_models/second.yaml) | 68.03/67.44	| 59.57/59.04 | 61.14/50.33	| 53.00/43.56 | 54.66/53.31 | 52.67/51.37 | 
| [Part-A^2-Anchor](tools/cfgs/waymo_models/PartA2.yaml) | 71.82/71.29 | 64.33/63.82 | 63.15/54.96 | 54.24/47.11 | 65.23/63.92 | 62.61/61.35 |
| [PV-RCNN](tools/cfgs/waymo_models/pv_rcnn.yaml) | 74.06/73.38 | 64.99/64.38 |	62.66/52.68 | 53.80/45.14 |	63.32/61.71	| 60.72/59.18 | 

My training runs trained **and** evaluated on 1/5th of the data:
|                                             | Vec_L1 | Vec_L2 | Ped_L1 | Ped_L2 | Cyc_L1 | Cyc_L2 |  
|---------------------------------------------|----------:|:-------:|:-------:|:-------:|:-------:|:-------:|
| [SECOND](tools/cfgs/waymo_models/second.yaml) | 67.85/67.24	| 59.35/58.80 | 57.94/47.85	| 49.93/41.18 | **55.54**/**54.12** | **53.75**/**52.38** | 
| [Part-A^2-Anchor](tools/cfgs/waymo_models/PartA2.yaml) | 71.57/71.03 | 64.08/63.57 | **63.34**/**55.08** | **54.39**/**47.20** | 64.93/63.71 | 62.53/**61.36** |
| [PV-RCNN](tools/cfgs/waymo_models/pv_rcnn.yaml) | **74.36**/**73.68** | **65.27**/**64.67** |	**63.29**/**53.04** | **54.36**/**45.46** |	**64.02**/**62.07**	| **61.53**/**59.66** | 

These are the deltas:
|                                             | Vec_L1 | Vec_L2 | Ped_L1 | Ped_L2 | Cyc_L1 | Cyc_L2 |  
|---------------------------------------------|----------:|:-------:|:-------:|:-------:|:-------:|:-------:|
| [SECOND](tools/cfgs/waymo_models/second.yaml) | -0.18/-0.20	| -0.22/-0.24 | -3.20/-2.48	| -3.07/-2.38 | **+0.88**/**+0.81** | **+1.08**/**+1.01** | 
| [Part-A^2-Anchor](tools/cfgs/waymo_models/PartA2.yaml) | -0.25/-0.26 | -0.25/-0.25 | **+0.19**/**+0.12** | **+0.15**/**+0.09** | -0.30/-0.21 | -0.08/**+0.01** |
| [PV-RCNN](tools/cfgs/waymo_models/pv_rcnn.yaml) | **+0.30**/**+0.30** | **+0.28**/**+0.31** |	**+0.63**/**+0.36** | **+0.56**/**+0.32** |	**+0.70**/**+0.36**	| **+0.81**/**+0.48** | 

The evaluation on the **full** validation set is still ongoing, but the trend goes in the same direction.
[image]
Depicted are two PV-RCNN runs (Step = Epoch).
Grey: trained and evaluated on 1/5th of the data.
Red: trained on 1/5th of the data, evaluation (ongoing) on the full validation set.

MartinHahner, 2021-01-23:
This is a comparison between an evaluation done on 1/5th of the validation set (in green)
and an evaluation done on the complete validation set (in blue and pink [continued])
of a [Part-A^2-Anchor](https://github.com/open-mmlab/OpenPCDet/issues/tools/cfgs/waymo_models/PartA2.yaml) training run on 1/5th of the training set.

So, in the end, there is almost no difference between the two final numbers.

[image]

w111liang222, 2021-03-05:
> I want to confirm (most of) the numbers given in [README.md](https://github.com/open-mmlab/OpenPCDet/blob/master/README.md).
> 
> Numbers provided by OpenPCDet on the full evaluation set:
> 
> Vec_L1	Vec_L2	Ped_L1	Ped_L2	Cyc_L1	Cyc_L2
> [SECOND](tools/cfgs/waymo_models/second.yaml)	68.03/67.44	59.57/59.04	61.14/50.33	53.00/43.56	54.66/53.31	52.67/51.37
> [Part-A^2-Anchor](tools/cfgs/waymo_models/PartA2.yaml)	71.82/71.29	64.33/63.82	63.15/54.96	54.24/47.11	65.23/63.92	62.61/61.35
> [PV-RCNN](tools/cfgs/waymo_models/pv_rcnn.yaml)	74.06/73.38	64.99/64.38	62.66/52.68	53.80/45.14	63.32/61.71	60.72/59.18
> My training runs trained **and** evaluated on 1/5th of the data:
> 
> Vec_L1	Vec_L2	Ped_L1	Ped_L2	Cyc_L1	Cyc_L2
> [SECOND](tools/cfgs/waymo_models/second.yaml)	67.85/67.24	59.35/58.80	57.94/47.85	49.93/41.18	**55.54**/**54.12**	**53.75**/**52.38**
> [Part-A^2-Anchor](tools/cfgs/waymo_models/PartA2.yaml)	71.57/71.03	64.08/63.57	**63.34**/**55.08**	**54.39**/**47.20**	64.93/63.71	62.53/**61.36**
> [PV-RCNN](tools/cfgs/waymo_models/pv_rcnn.yaml)	**74.36**/**73.68**	**65.27**/**64.67**	**63.29**/**53.04**	**54.36**/**45.46**	**64.02**/**62.07**	**61.53**/**59.66**
> These are the deltas:
> 
> Vec_L1	Vec_L2	Ped_L1	Ped_L2	Cyc_L1	Cyc_L2
> [SECOND](tools/cfgs/waymo_models/second.yaml)	-0.18/-0.20	-0.22/-0.24	-3.20/-2.48	-3.07/-2.38	**+0.88**/**+0.81**	**+1.08**/**+1.01**
> [Part-A^2-Anchor](tools/cfgs/waymo_models/PartA2.yaml)	-0.25/-0.26	-0.25/-0.25	**+0.19**/**+0.12**	**+0.15**/**+0.09**	-0.30/-0.21	-0.08/**+0.01**
> [PV-RCNN](tools/cfgs/waymo_models/pv_rcnn.yaml)	**+0.30**/**+0.30**	**+0.28**/**+0.31**	**+0.63**/**+0.36**	**+0.56**/**+0.32**	**+0.70**/**+0.36**	**+0.81**/**+0.48**
> The evaluation on the **full** validation set is still ongoing, but the trend goes in the same direction.
> [image]
> Depicted are two [PV-RCNN](tools/cfgs/waymo_models/pv_rcnn.yaml) runs (Step = Epoch).
> Grey: trained and evaluated on 1/5th of the data.
> Red: trained on 1/5th of the data, evaluation (ongoing) on the full validation set.
@MartinHahner 
I trained the SECOND with the default configuration and test the model on the whole validation, and I got the similar results with yours. The numbers of PED L1/L2 are lower than the numbers given in README.md about 3%. Did you find out the problem? Thanks.

MartinHahner, 2021-03-05:
If you trained SECOND once and you observe this, I think it is within the "normal" variance of different training runs. 
If you let's say trained SECOND five times, and every time it is by 3% lower than the reported numbers, then I don't know why.

HenryJunW, 2021-03-15:
> I want to confirm (most of) the numbers given in [README.md](https://github.com/open-mmlab/OpenPCDet/blob/master/README.md).
>
> Numbers provided by OpenPCDet on the full evaluation set:
> 	Vec_L1 [...]

Thanks for sharing the reproduce results. How many GPUs are you using to reproduce the pv_rcnn results? Thanks.

MartinHahner, 2021-03-15:
I think I used 4 or 8 GPUs. This shouldn't matter though, it would only have an impact on the training time.

HenryJunW, 2021-03-15:
Thanks for your reply! Since I am trying to reproduce pv-rcnn results on 4 P6000 GPUs, but I am getting ~0.61 at epoch 24, which is way below what you get. I am wondering what the problem could be. Are you using the default evaluation code w/o anything modification? https://github.com/open-mmlab/OpenPCDet/blob/0642cf06d0fd84f50cc4c6c01ea28edbc72ea810/pcdet/datasets/waymo/waymo_eval.py#L239 indicates the dist_threshold to be 1000, so I am not sure whether you change it to be 30/50 to be consistent with the Waymo official metric. Thank you!

MartinHahner, 2021-03-15:
Nope, I did not change anything.

HenryJunW, 2021-03-16:
Thanks for your information!

xuanzhangyang, 2021-03-30:
training pv-rcnn on waymo dataset:

/home/xuan/code/pytorch/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:76: RuntimeWarning: overflow encountered in multiply03:41,  2.71s/it, total_it=89722]
  points[:, :3] *= noise_scale
epochs:  90%|██████████████████████████████████████████████████████████████████████████████▎        | 27/30 [70:45:38<7:52:40, 9453.55s/it, loss=1.66, lr=0.000627]/home/xuan/code/pytorch/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:76: RuntimeWarning: overflow encountered in multiply19:00,  2.68s/it, total_it=93601]
  points[:, :3] *= noise_scale
epochs:  93%|█████████████████████████████████████████████████████████████████████████████████▏     | 28/30 [73:22:38<5:15:11, 9455.94s/it, loss=1.56, lr=0.000274]/home/xuan/code/pytorch/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:76: RuntimeWarning: overflow encountered in multiply20:07,  2.69s/it, total_it=97040]
  points[:, :3] *= noise_scale
epochs:  93%|█████████████████████████████████████████████████████████████████████████████████▏     | 28/30 [74:20:43<5:15:11, 9455.94s/it, loss=2.27, lr=0.000178]/home/xuan/code/pytorch/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py:76: RuntimeWarning: overflow encountered in multiply21:23,  2.66s/it, total_it=98332]
  points[:, :3] *= noise_scale
train: 100%|████████████████████████████████████████████████████████████████████████████████████████████████| 3454/3454 [2:35:10<00:00,  2.70s/it, total_it=103620]
epochs: 100%|█████████████████████████████████████████████████████████████████████████████████████████████| 30/30 [78:20:16<00:00, 9400.56s/it, loss=2.79, lr=1e-7]
2021-03-30 17:18:14,818   INFO  **********************End training waymo_models/pv_rcnn(default)**********************

### epoch 30: 

2021-03-30 19:22:49,337   INFO  recall_roi_0.3: 0.613171
2021-03-30 19:22:49,337   INFO  recall_rcnn_0.3: 0.613844
2021-03-30 19:22:49,337   INFO  recall_roi_0.5: 0.550942
2021-03-30 19:22:49,337   INFO  recall_rcnn_0.5: 0.562655
2021-03-30 19:22:49,337   INFO  recall_roi_0.7: 0.308685
2021-03-30 19:22:49,337   INFO  recall_rcnn_0.7: 0.365962

OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.5073
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.4977
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.4482
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.4395
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.4421
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.3412
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.3704
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.2857
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.3261
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.3033
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.3154
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.2933

closed 2022-03-08 by github-actions[bot]

r-sy, 2022-03-19:
> 
Hi, excuse me，`waymo_processed_data_v0_5_0/segment-10017090168044687777_6380_000_6400_000_with_camera_labels/0000.npy` generated by `waymo-open-dataset-tf-2-5-0` in the latest version v0.5.2 is still the same as before? 
However,  load it with the following line:
```
A = np.load('0000.npy')
print(A.shape)
print(A.sum())
```
I get the results are (183680, 6) and 586412.5. Is there something wrong with my generation? 

Is this the reason why I can't fully reproduce the accuracy of PV_RCNN and other algorithms on Waymo dataset? Three or four points lower on average. 

I hope you can get a reply and the problem can be solved. Thank you!

I get pv_rcnn : 
```
2022-03-13 23:23:58,300   INFO  *************** EPOCH 30 EVALUATION *****************
2022-03-13 23:52:17,842   INFO  *************** Performance of EPOCH 30 *****************
2022-03-13 23:52:17,858   INFO  Generate label finished(sec_per_example: 0.0425 second).
2022-03-13 23:52:17,858   INFO  recall_roi_0.3: 0.804661
2022-03-13 23:52:17,858   INFO  recall_rcnn_0.3: 0.805084
2022-03-13 23:52:17,858   INFO  recall_roi_0.5: 0.741911
2022-03-13 23:52:17,858   INFO  recall_rcnn_0.5: 0.751780
2022-03-13 23:52:17,859   INFO  recall_roi_0.7: 0.486838
2022-03-13 23:52:17,859   INFO  recall_rcnn_0.7: 0.533652
2022-03-13 23:52:17,884   INFO  Average predicted number of objects(39987 samples): 88.996
2022-03-14 00:03:08,822   INFO  
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.7297 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.7227 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.6441 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.6377 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.6967 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.5787 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.6053 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.5013 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.6462 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.6244 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.6222 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.6013
``` 
pv_rcnn_center: 
```
2022-03-19 12:57:09,953   INFO  **********************Start evaluation waymo_models/pv_rcnn_with_centerhead_rpn(default)**********************
2022-03-19 12:57:09,955   INFO  Loading Waymo dataset
2022-03-19 12:57:22,970   INFO  Total skipped info 0
2022-03-19 12:57:22,971   INFO  Total samples for Waymo dataset: 39987
2022-03-19 12:57:22,973   INFO  ==> Loading parameters from checkpoint ./OpenPCDet/output/waymo_models/pv_rcnn_with_centerhead_rpn/default/ckpt/checkpoint_epoch_30.pth to CPU
2022-03-19 12:57:23,098   INFO  ==> Checkpoint trained from version: pcdet+0.5.2+a9c66fe
2022-03-19 12:57:23,154   INFO  ==> Done (loaded 437/437)
2022-03-19 12:57:23,158   INFO  *************** EPOCH 30 EVALUATION *****************
2022-03-19 13:28:13,964   INFO  *************** Performance of EPOCH 30 *****************
2022-03-19 13:28:14,026   INFO  Generate label finished(sec_per_example: 0.0463 second).
2022-03-19 13:28:14,026   INFO  recall_roi_0.3: 0.834954
2022-03-19 13:28:14,026   INFO  recall_rcnn_0.3: 0.835461
2022-03-19 13:28:14,027   INFO  recall_roi_0.5: 0.762491
2022-03-19 13:28:14,027   INFO  recall_rcnn_0.5: 0.772247
2022-03-19 13:28:14,027   INFO  recall_roi_0.7: 0.473409
2022-03-19 13:28:14,027   INFO  recall_rcnn_0.7: 0.532824
2022-03-19 13:28:14,058   INFO  Average predicted number of objects(39987 samples): 86.808
2022-03-19 13:43:13,291   INFO  
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.7393 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.7338 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.6538 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.6488 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.7248 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.6550 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.6359 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.5732 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.6900 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.6762 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.6646 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.6513 
```

sshaoshuai, 2022-03-19:
@r-sy , 
Hi, the data has been updated after `version 0.5` and your current data state is consistent with me. 
So you may check other reasons to match the performance.

r-sy, 2022-03-23:
Ok，thank you very much！---- Replied Message ----FromShaoshuai ***@***.***>Date03/19/2022 20:22 ***@***.***> ***@***.******@***.***>SubjectRe: [open-mmlab/OpenPCDet] Low accuracy when training on waymo dataset (#370)
@r-sy ,
Hi, the data has been updated after version 0.5 and your current data state is consistent with me.
So you may check other reasons to match the performance.

[
{
"target": "https://github.com/open-mmlab/OpenPCDet/issues/370#issuecomment-1073000369",
"url": "https://github.com/open-mmlab/OpenPCDet/issues/370#issuecomment-1073000369",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

jwma0725, 2022-04-29:
> @r-sy , Hi, the data has been updated after `version 0.5` and your current data state is consistent with me. So you may check other reasons to match the performance.
@sshaoshuai 
Hi, I also use the new version 0.5 to generate the waymo dataset(waymo version 1.0) with waymo-open-dataset-tf-2-5-0-1.4.1. But I do not obtain the same result.  
I think the problem is the numpy version. The waymo-open-dataset-tf-2-5-0 need numpy==1.19.5.  I statisfy this version, but occur error: "ImportError: numpy.core.multiarray failed to import". And this error comes from "import SharedArray" . Based on this error, I update numpy version to 1.20, this error does not occur.  After I trained Voxel-RCNN and val, another error occur:"can not convert a symbolic Tensor(strided_slice:0) to a numpy array. This error may indicate that you're trying to pass a Tensor to a Numpy call. which is not supported" . I return to the origninal numpy version==1.19.5 and try to delete these code about "SharedArray", the code could run but all the result are zero.(Start the waymo evaluation... Number:(pd, 1322657)VS. (gt,0)) also generates. Can you give me some advice on these problems? Thank u very much!!!

jwma0725, 2022-04-29:
@sshaoshuai 
Hi, I also use the new version 0.5 to generate the waymo dataset(waymo version 1.0) with waymo-open-dataset-tf-2-5-0-1.4.1. But I do not obtain the same result.  
I think the problem is the numpy version. The waymo-open-dataset-tf-2-5-0 need numpy==1.19.5.  I statisfy this version, but occur error: "ImportError: numpy.core.multiarray failed to import". And this error comes from "import SharedArray" . Based on this error, I update numpy version to 1.20, this error does not occur.  After I trained Voxel-RCNN and val, another error occur:"can not convert a symbolic Tensor(strided_slice:0) to a numpy array. This error may indicate that you're trying to pass a Tensor to a Numpy call. which is not supported" . I return to the origninal numpy version==1.19.5 and try to delete these code about "SharedArray", the code could run but all the result are zero.(Start the waymo evaluation... Number:(pd, 1322657)VS. (gt,0)) also generates. Can you give me some advice on these problems? Thank u very much!!!

jwma0725, 2022-04-29:
@r-sy Hi, I also use waymo-open-dataset-tf-2-5-0 and the numpy version is 1.19.5. But I do not run code nomally and generate this error:"ImportError: numpy.core.multiarray failed to import". Can u tell me your numpy version or other suggestions? Thank u very much!!!

r-sy, 2022-05-05:
> @r-sy Hi, I also use waymo-open-dataset-tf-2-5-0 and the numpy version is 1.19.5. But I do not run code nomally and generate this error:"ImportError: numpy.core.multiarray failed to import". Can u tell me your numpy version or other suggestions? Thank u very much!!!

You can try to install SharedArray==3.1.0.

jwma0725, 2022-05-06:
Thanks!!! It works well!

carrolltm, 2023-05-26:
> 

> 
Hi, I also have this problem when I use waymo v1.0 for object detection(Start the waymo evaluation... Number:(pd, 1322657)VS. (gt,0))，i also use the environment，Can you tell me how you finally solved it?Thank u very much!!!
[image]

RookieXwc, 2023-10-12:
> You can try to install SharedArray==3.1.0.

I have the same issue with jwma0725, and install SharedArray==3.1.0 work!!!!!!!