open-mmlab/OpenPCDet #376: Training  error on other dataset

chenxyyy, 2020-11-25:
Hi!  Thank you for your work. I like the point cloud detection frame.

I want to use PV_RCNN algorithm in this framework to train my own dataset, but I encountered the following error. How can I solve it?

My own dataset are labeled by myself, which is labeled on raw pointcloud file. I have modify the code about datalod. when I debuge this procejt on 
 ```
> /data/OpenPCDet/pcdet/models/__init__.py(30) model_func() 
> ret_dict, tb_dict, disp_dict = model(batch_dict)                                                                                                       ```
```
I have confirmed that my input batch_dict are as follows

```
(Pdb) batch_dict.keys()
dict_keys(['points', 'frame_id', 'gt_boxes', 'use_lead_xyz', 'voxels', 'voxel_coords', 'voxel_num_points', 'image_shape', 'batch_size'])
``` 
that is same as your input when traing on kitti dataset.

But I don't know why I met this error. Could you help me with this problem? 

Thank you very!
 
 I look forward to your reply. 

```
Traceback (most recent call last):
  File "train.py", line 199, in <module>
    main()
  File "train.py", line 171, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/data/OpenPCDet/tools/train_utils/train_utils.py", line 95, in train_model
    dataloader_iter=dataloader_iter
  File "/data/OpenPCDet/tools/train_utils/train_utils.py", line 39, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/data/OpenPCDet/pcdet/models/__init__.py", line 30, in model_func
    load_data_to_gpu(batch_dict)
  File "/root/anaconda3/lib/python3.7/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/data/OpenPCDet/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/root/anaconda3/lib/python3.7/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/data/OpenPCDet/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 235, in forward
    point_features = self.vsa_point_feature_fusion(point_features.view(-1, point_features.shape[-1]))
  File "/root/anaconda3/lib/python3.7/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/root/anaconda3/lib/python3.7/site-packages/torch/nn/modules/container.py", line 92, in forward
    input = module(input)
  File "/root/anaconda3/lib/python3.7/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/root/anaconda3/lib/python3.7/site-packages/torch/nn/modules/linear.py", line 92, in forward
    return F.linear(input, self.weight, self.bias)
  File "/root/anaconda3/lib/python3.7/site-packages/torch/nn/functional.py", line 1408, in linear
    output = input.matmul(weight.t())
RuntimeError: size mismatch, m1: [2048 x 768], m2: [640 x 128] at /tmp/pip-req-build-pb3z3zl3/aten/src/THC/generic/THCTensorMathBlas.cu:268

```

sshaoshuai, 2020-11-28:
It seems you have modified the configs of VSA layer, so the size doesn't match. 
You may check your config by comparing with my original config.

chenxyyy, 2020-11-30:
Hi! @sshaoshuai 

I have solved this error. This error is due to the ratio of detection range to voxel.

Your detection range is x[0,70.4], y[-40, 40], z[-3, 1]. when I train my 32line lidar dataset, I have change it to [-70.4, -80, -6, 70.4, 80, 2], which is x[-70.4, 70.4], y[-80, 80], z[-6, 2]. Compared with your algorithm detection range, I expanded by two times. But I didn't chage the voxel of your config of [0.05, 0.05, 0.1]. So I chage it to [0.1, 0.1, 0.2], this problem disappears.

chenxyyy, 2020-11-30:
The pvrcnn algorithm works very well. but there are some problems. You can see my test image on my blog about 
[pcdet test](https://blog.csdn.net/weixin_39446416/article/details/110368320)，

As you can see, the heading angles of some vehicles are offset, but the heading angles of the remaining vehicles are very accurate. Also, vehicles with wrong heading angles seem to have a certain pattern.

Do you know why this happens？ @sshaoshuai 

Thank you very much！

closed 2020-12-07 by chenxyyy

cowarder, 2020-12-09:
> The pvrcnn algorithm works very well. but there are some problems. You can see my test image on my blog about
> [pcdet test](https://blog.csdn.net/weixin_39446416/article/details/110368320)，
>
> As you [...]

Hi, I can't open the link of your bad case, could you check the link?

chenxyyy, 2020-12-10:
@cowarder Hi!

Thank you for your attention! I've solved this problem.

This is not a problem with your code, but the angle definition of my dataset is different from that of Kitti. After modifying the angle, I retrained the model. The effect was very good, so I deleted the blog. Thank you for your attention and your great work!

cowarder, 2020-12-14:
@chenxyyy I met the same problem as you when training with custom pcd data, I change the detection range as you mentioned. But when I run demo.py to check the model performance, the fig is always black and show nothing. Any other special configs needs to tune when you run demo.py?

chenxyyy, 2020-12-14:
What you're describing is that the model that you've trained doesn't have any output when inference, or it does have output but it's a visual error?

cowarder, 2020-12-15:
[image]
The detection result looks like this. It seems to be incomplete, and only part was shown in this pic?

ChunmianLin, 2021-12-10:
@chenxyyy Hi, I also encounter some problem when I train PV-RCNN on my own pcd data, and I guess it is caused by the angle definition. The coordinate of my own dataset is the same with local coordinate defined by PV-RCNN, apart from clockwise transformation. Could you please describe your data case and how you modify the angle to adapt the code?
Thanks in advnace.