open-mmlab/OpenPCDet #378: PC crashed when running python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos

justinchu1997, 2020-11-25:
when running python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos

cpu got high utilization and then  the pc crashed

chenxyyy, 2020-11-26:
the code 
```
python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos
```
 are actually five tasks running, five files were generated respectively. 

You can split them into four tasks, one at a time. You can split them into four tasks, one at a time. You just need to comment out the others.

In this way, the computer won't crashed

```
print('---------------Start to generate data infos---------------')
# first
dataset.set_split(train_split)
kitti_infos_train = dataset.get_infos(num_workers=workers, has_label=True, count_inside_pts=True)
with open(train_filename, 'wb') as f:
    pickle.dump(kitti_infos_train, f)
print('Kitti info train file is saved to %s' % train_filename)

# second
# dataset.set_split(val_split)
# kitti_infos_val = dataset.get_infos(num_workers=workers, has_label=True, count_inside_pts=True)
# with open(val_filename, 'wb') as f:
#     pickle.dump(kitti_infos_val, f)
# print('Kitti info val file is saved to %s' % val_filename)

# third
# with open(trainval_filename, 'wb') as f:
#     pickle.dump(kitti_infos_train + kitti_infos_val, f)
# print('Kitti info trainval file is saved to %s' % trainval_filename)
# pdb.set_trace()

# fourth
# dataset.set_split('test')
# kitti_infos_test = dataset.get_infos(num_workers=workers, has_label=False, count_inside_pts=False)
# # pdb.set_trace()
# with open(test_filename, 'wb') as f:
#     pickle.dump(kitti_infos_test, f)
# print('Kitti info test file is saved to %s' % test_filename)

# fifth
# print('---------------Start create groundtruth database for data augmentation---------------')
# dataset.set_split(train_split)
# pdb.set_trace()
# dataset.create_groundtruth_database(train_filename, split=train_split)

print('---------------Data preparation Done---------------')
```

justinchu1997, 2020-11-26:
Splitting data preparation into  five tasks didn't work.
I modify the get_infos function and split all the  points into 8 slices.
This works for me.
`def get_infos(self, num_workers=4, has_label=True, count_inside_pts=True, sample_id_list=None):
        import concurrent.futures as futures

        def process_single_scene(sample_idx):
            print('%s sample_idx: %s' % (self.split, sample_idx))
            info = {}
            pc_info = {'num_features': 4, 'lidar_idx': sample_idx}
            info['point_cloud'] = pc_info
            
            image_info = {'image_idx': sample_idx, 'image_shape': self.get_image_shape(sample_idx)}
            info['image'] = image_info
            calib = self.get_calib(sample_idx)
                
            P2 = np.concatenate([calib.P2, np.array([[0., 0., 0., 1.]])], axis=0)
            R0_4x4 = np.zeros([4, 4], dtype=calib.R0.dtype)
            R0_4x4[3, 3] = 1.
            R0_4x4[:3, :3] = calib.R0
            V2C_4x4 = np.concatenate([calib.V2C, np.array([[0., 0., 0., 1.]])], axis=0)
            calib_info = {'P2': P2, 'R0_rect': R0_4x4, 'Tr_velo_to_cam': V2C_4x4}

            info['calib'] = calib_info

            if has_label:
                obj_list = self.get_label(sample_idx)
                annotations = {}
                annotations['name'] = np.array([obj.cls_type for obj in obj_list])
                annotations['truncated'] = np.array([obj.truncation for obj in obj_list])
                annotations['occluded'] = np.array([obj.occlusion for obj in obj_list])
                annotations['alpha'] = np.array([obj.alpha for obj in obj_list])
                annotations['bbox'] = np.concatenate([obj.box2d.reshape(1, 4) for obj in obj_list], axis=0)
                annotations['dimensions'] = np.array([[obj.l, obj.h, obj.w] for obj in obj_list])  # lhw(camera) format
                annotations['location'] = np.concatenate([obj.loc.reshape(1, 3) for obj in obj_list], axis=0)
                annotations['rotation_y'] = np.array([obj.ry for obj in obj_list])
                annotations['score'] = np.array([obj.score for obj in obj_list])
                annotations['difficulty'] = np.array([obj.level for obj in obj_list], np.int32)

                num_objects = len([obj.cls_type for obj in obj_list if obj.cls_type != 'DontCare'])
                num_gt = len(annotations['name'])
                index = list(range(num_objects)) + [-1] * (num_gt - num_objects)
                annotations['index'] = np.array(index, dtype=np.int32)

                loc = annotations['location'][:num_objects]
                dims = annotations['dimensions'][:num_objects]
                rots = annotations['rotation_y'][:num_objects]
                loc_lidar = calib.rect_to_lidar(loc)
                l, h, w = dims[:, 0:1], dims[:, 1:2], dims[:, 2:3]
                loc_lidar[:, 2] += h[:, 0] / 2
                gt_boxes_lidar = np.concatenate([loc_lidar, l, w, h, -(np.pi / 2 + rots[..., np.newaxis])], axis=1)
                annotations['gt_boxes_lidar'] = gt_boxes_lidar

                info['annos'] = annotations

                if count_inside_pts:
                    points = self.get_lidar(sample_idx)
                    calib = self.get_calib(sample_idx)
                    pts_rect = np.zeros((points.shape[0],3))
                    fov_flag = np.zeros(points.shape[0],dtype='B')
                    num_split = 8
                    for i in range(num_split):
                        begin = i*int(points.shape[0]/num_split)
                        end = min(points.shape[0],(i+1)*int(points.shape[0]/num_split))
                        if i == num_split-1:
                            end = points.shape[0]
                        pts_rect[begin:end] = calib.lidar_to_rect(points[begin:end,0:3])
                        fov_flag[begin:end] = self.get_fov_flag(pts_rect[begin:end], info['image']['image_shape'], calib)
                    pts_fov = points[fov_flag]
                    corners_lidar = box_utils.boxes_to_corners_3d(gt_boxes_lidar)
                    num_points_in_gt = -np.ones(num_gt, dtype=np.int32)

                    for k in range(num_objects):
                        flag = box_utils.in_hull(pts_fov[:, 0:3], corners_lidar[k])
                        num_points_in_gt[k] = flag.sum()
                    annotations['num_points_in_gt'] = num_points_in_gt
            return info

        sample_id_list = sample_id_list if sample_id_list is not None else self.sample_id_list
        with futures.ThreadPoolExecutor(max_workers=num_workers) as executor:
            infos = executor.map(process_single_scene,sample_id_list)
        return list(infos)`

sshaoshuai, 2020-11-28:
I didn't have this problem and my PC just has 16G cpu memory

closed 2020-11-30 by justinchu1997