open-mmlab/OpenPCDet #380: What's the better way to normalize intensity？

a43992899, 2020-11-25:
I see the author use tanh to normalize intensity in waymo, but tanh can be easily saturated, making most of the values close to 1. Is it a good way to do this? Or what's the better way to normalize this feature？Simply divide it by 255？
Also, I find that model train on kitti cannot generalize well on my dataset, but if I set my intensity to 0, the pretrain model can achieve better result on my dataset. Is it a hardware-dependent feature？

tianweiy, 2020-11-25:
Simply divide it by 255 <- it is not between 0~255 for Waymo intensity. It can go up to 40k so tanh seems to be the normal solution. 

Is it a hardware-dependent feature <- yes

a43992899, 2020-11-26:
Thank you for your help. @tianweiy

closed 2020-11-26 by a43992899

MartinHahner, 2020-11-26:
> I find that the pretrain kitti model cannot generalize well on my dataset

I also made this observation

> but if I set my intensity to 0, the pretrain model can achieve better result on my dataset.

That is interesting! Thanks for sharing. :+1:

MartinHahner, 2020-11-26:
@a43992899 Did you also use 0 intensity for your own data when you finetuned from the 0 intensity KITTI checkpoints?

a43992899, 2020-11-26:
@MartinHahner I do set my intensity to 0. But I don't fine-tuned the Kitti model. When I find that the KITTI pretrain model doesn't generalize well, I retrain a new model on my dataset. Another reason is that I have lidar data from many types of hardware, since the intensity is a hardware-dependent feature, maybe I should not use the intensity at all?

MartinHahner, 2020-11-26:
@a43992899 Ah OK, then I understood you wrong. 
Still, when you transfer from one dataset to another one that was not collected with exacltly the same sensor setup, it might be better to not use the intensity at all. I will try this on my end.

HuangVictorAuto, 2021-09-07:
@a43992899 , which method did you use later, to use the intesity or not and why? can you share some information, thank you!

Murdism, 2024-06-03:
> > I find that the pretrain kitti model cannot generalize well on my dataset
> 
> I also made this observation.
> 
> > but if I set my intensity to 0, the pretrain model can achieve better result on my dataset.
> 
> That is interesting! Thanks for sharing. 👍

Did you retrain on your new dataset without intensity or use pretrained model with intensity set to zero.