open-mmlab/OpenPCDet #382: Strange result for nuscenes dataset

yeyang1021, 2020-11-27:
We only use 2 sweeps to train the model (SECOND-MultiHead (CBGS)). 
There is only the object detection result of the car.

The result as below:

2020-11-19 02:38:40,873   INFO  ----------------Nuscene detection_cvpr_2019 results-----------------
***car error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.19, 0.15, 0.15, 0.52, 0.23 | 66.50, 77.35, 81.94, 84.01 | mean AP: 0.7744819851380316
***truck error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, 1.00, 1.00 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***construction_vehicle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, 1.00, 1.00 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***bus error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, 1.00, 1.00 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***trailer error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, 1.00, 1.00 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***barrier error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, nan, nan | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***motorcycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, 1.00, 1.00 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***bicycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, 1.00, 1.00 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***pedestrian error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, 1.00, 1.00, 1.00 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***traffic_cone error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
1.00, 1.00, nan, nan, nan | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
--------------average performance-------------
trans_err:	 0.9193
scale_err:	 0.9152
orient_err:	 0.9056
vel_err:	 0.9401
attr_err:	 0.9033
mAP:	 0.0774
NDS:	 0.0804

2020-11-19 02:38:40,874   INFO  Result is save to /home/yeyang/Detectors/PCDet/OpenPCDet/output/cfgs/nuscenes_models/cbgs_second_multihead/default/eval/eval_with_train/epoch_20/val
2020-11-19 02:38:40,874   INFO  ****************Evaluation done.*****************
2020-11-19 02:38:40,900   INFO  Epoch 20 has been evaluated
2020-11-19 02:39:10,933   INFO  **********************End evaluation cfgs/nuscenes_models/cbgs_second_multihead(default)**********************

sshaoshuai, 2020-11-28:
Have you tried to use 10 sweeps to train it? 
How do you modify the codes or configs?

yeyang1021, 2020-12-01:
> Have you tried to use 10 sweeps to train it?
> How do you modify the codes or configs?

I re-downloaded a copy of the code and compiled it, but the training result still had the above problem.
The same problem was encountered with 10 sweeps. But there is no problem using the test model from your.

w111liang222, 2020-12-01:
> > Have you tried to use 10 sweeps to train it?
> > How do you modify the codes or configs?
> 
> I re-downloaded a copy of the code and compiled it, but the training result still had the above problem.
> The same problem was encountered with 10 sweeps. But there is no problem using the test model from your.

I had the problem too, After looking into the code, I find the commit 1145f836f9091c7e6fa1fb825e377f5efac134cc causing the problem.
In the commit, The target assigner set the SEPARATE_MULTIHEAD to True, then it remaps the label for 0 to  rpn_class_num
But, when calculating the cls_layer_loss, the function gets the one_hot_target is wrong:
one_hot_target = one_hot_targets[:, start_idx:start_idx + cls_pred.shape[1],
                                 c_idx:c_idx + cur_num_class]
the c_idx is increased by cur_num_class (rpn_class_num)

yeyang1021, 2020-12-02:
> > > Have you tried to use 10 sweeps to train it?
> > > How do you modify the codes or configs?
> > 
> > 
> > I re-downloaded a copy of the code and compiled it, but the training result still had the above problem.
> > The same problem was encountered with 10 sweeps. But there is no problem using the test model from your.
> 
> I had the problem too, After looking into the code, I find the commit [1145f83](https://github.com/open-mmlab/OpenPCDet/commit/1145f836f9091c7e6fa1fb825e377f5efac134cc) causing the problem.
> In the commit, The target assigner set the SEPARATE_MULTIHEAD to True, then it remaps the label for 0 to rpn_class_num
> But, when calculating the cls_layer_loss, the function gets the one_hot_target is wrong:
> one_hot_target = one_hot_targets[:, start_idx:start_idx + cls_pred.shape[1],
> c_idx:c_idx + cur_num_class]
> the c_idx is increased by cur_num_class (rpn_class_num)

Do you solve this problem?

sshaoshuai, 2020-12-02:
@yeyang1021 @w111liang222 ,  thank you for the bug report. I have fixed it in https://github.com/open-mmlab/OpenPCDet/pull/392.

closed 2020-12-02 by sshaoshuai