open-mmlab/OpenPCDet #385: waymo preprocessing numbers

MartinHahner, 2020-11-27:
@sshaoshuai 
Can you please confirm if those numbers are the way they should be now? 

training:
```
2020-11-27 21:09:16,923   INFO  Loading Waymo dataset
2020-11-27 21:09:23,159   INFO  Total skipped info 0
2020-11-27 21:09:23,159   INFO  Total samples for Waymo dataset: 158081
2020-11-27 21:09:23,170   INFO  Total sampled samples for Waymo dataset: 31617
```

validation: 
```
2020-11-27 21:09:14,988   INFO  Loading Waymo dataset
2020-11-27 21:09:16,644   INFO  Total skipped info 0
2020-11-27 21:09:16,644   INFO  Total samples for Waymo dataset: 39987
2020-11-27 21:09:16,646   INFO  Total sampled samples for Waymo dataset: 7998
```

P.S. It again took multiple days for me to generate them. 
This time there were no errors, but the script was killed multiple times (probably due to too much memory consumption as I read in https://github.com/open-mmlab/OpenPCDet/issues/378)

sshaoshuai, 2020-11-28:
Yes, these numbers are correct for the version 1.2 of Waymo dataset. 
Note that you just need to re-generate validation infos for evaluating with Waymo metrics since previous training data and infos are correct. 

I am not sure why it needs multiple days for generating them, and my codes don't have computational intensively parts. Maybe it is due to the data decoding with tensorflow apis, and sometimes the data processing is fast and sometimes it is slow, and I guess it depends on the environment of tensorflow.

closed 2020-11-28 by MartinHahner

jialeli1, 2021-01-13:
Hi, how long is the normal processing time of waymo dataset?
I am preparing the waymo dataset, but I estimated that it will take almost 30+ hours. Is this time consumption reasonable? Or maybe there are some problems with my tensorflow environment?

jihanyang, 2021-01-14:
@jialeli1 It is reasonable. We take around 40 hours.