open-mmlab/OpenPCDet #388: New Dataset - 'USE_ROAD_PLANE: False' Causes Train Error

mcboyd-bu, 2020-11-30:
Hello. I am training on simulation data from the Carla simulator. I used the Kitti and Waymo dataset configurations as my references, though I generally followed Waymo more closely. Trying to train, I get the following error:

...File "/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 125, in add_sampled_boxes_to_scene
    sampled_gt_boxes, data_dict['road_plane'], data_dict['calib']
KeyError: 'road_plane'

Because I have 3D box labels in my Lidar data (as Waymo does), I followed the Waymo "infos" structure, which means I left out the 'calib' data and the 'road_plane' data and functions. I also set my carla_dataset.yaml file to have "USE_ROAD_PLANE: False" (again, replicating waymo_dataset.yaml). 

The issue is the code in database_sampler.py (line 123) specifically looks for both 'calib' and 'road_plane' structures even with "USE_ROAD_PLANE: False":

 if self.sampler_cfg.get('USE_ROAD_PLANE', False):  
     sampled_gt_boxes, mv_height = self.put_boxes_on_road_planes(  
     sampled_gt_boxes, data_dict['road_plane'], data_dict['calib']  
            )  
            data_dict.pop('calib')  
            data_dict.pop('road_plane')

Should line 123 instead check for "USE_ROAD_PLANE" to be True? Are there other changes in database_sampler.py that were not uploaded when Waymo dataset support was added recently?

Thank you!
Matt

jihanyang, 2020-11-30:
Hello, it seems that your `self.sampler_cfg.get('USE_ROAD_PLANE', False)` is True? Could you please print the `self.sampler_cfg` before line 123 in database_sampler.py?

mcboyd-bu, 2020-11-30:
You're absolutely correct! There was a second "USE_ROAD_PLANE" setting that I missed in the pv_rcnn.yaml file that I copied from Kitti. Setting it to False moves me past this error. My apologies for bothering you with an oversight on my part. I appreciate your quick reply.

closed 2020-11-30 by mcboyd-bu