open-mmlab/OpenPCDet #389: results of the pre-trianed  model

hitxing, 2020-11-30:
Hi everyone! I've installed this projects and download the pretrained model  for  Second and PV-RCNN,  my problem is: 
when I running demo.py to  see the detected results on KITTI datasets(some samples of KITTI test sets ), the  results is very badly, there are a lot of  false detect results, then I replace the KITTI datasets with my own pointcloud bin file,  the performance is also very poor.
Did anyone occurred the same problem?

[image]
[image]

AymanNasser, 2021-01-11:
The confidence score of false detections are very low, around 20% so you can threshold the output to a specific score and you won't see these detections @hitxing 
I wanna ask you about something if you could help, when I load the pretrained weights for pv_rcnn and try to run demo.py, it outputs segmentation fault (core dumped), have you faced this issue ?

hitxing, 2021-01-11:
Thanks for your help, I faced the same problem when I trying to train my own weights but not in the time I load the pretrained weights,I tried some solutions but all failed,finally,I train the weights on another computer and it works.Hope I was helpful!

AngYikHong, 2021-02-23:
Hi @AymanNasser, may I know how did you change the confidence score of false detection?

AymanNasser, 2021-02-23:
@AngYikHong  In the output prediction dictionary `pred_dicts, _ = model.forward(data_dict)` in tools/demo **line 91**, you can access the score of each box by `pred_dicts[0]['pred_scores']` and threshold the visualization with a specific score threshold

closed 2022-03-08 by github-actions[bot]