open-mmlab/OpenPCDet #391: Error with demo script

harris-mohamed, 2020-12-01:
Hello, 

I am setting up OpenPCDet for the first time. I am trying to run the demo script as specified in DEMO.md, but am getting the error:
```
  File "demo.py", line 31
    data_file_list = glob.glob(str(root_path / f'*{self.ext}')) if self.root_path.is_dir() else [self.root_path]
                                                            ^
SyntaxError: invalid syntax
```
Does anyone have any idea what is causing this? Any help would be greatly appreciated. Thanks!

closed 2021-01-04 by harris-mohamed

turtleMaster20, 2021-05-07:
Hi, 

I am having the same issue, how did you solve it?

harris-mohamed, 2021-05-07:
If I remember correctly, I had the wrong version of spconv set up. You have to use the version specified in the readme.