open-mmlab/OpenPCDet #395: run demo.py

liu52020, 2020-12-03:
[image]
Hi,sir! Can you help me? Please!

liu52020, 2020-12-03:
2020-12-03 21:24:50.143 (   9.318s) [        46F83740]vtkOpenGLRenderWindow.c:582    ERR| vtkXOpenGLRenderWindow (0x5604fd8387f0): Unable to find a valid OpenGL 3.2 or later implementation. Please update your video card driver to the latest version. If you are using Mesa please make sure you have version 11.2 or later and make sure your driver in Mesa supports OpenGL 3.2 such as llvmpipe or openswr. If you are on windows and using Microsoft remote desktop note that it only supports OpenGL 3.2 with nvidia quadro cards. You can use other remoting software such as nomachine to avoid this issue.

shiqi1994, 2021-01-18:
May I ask your OpenGL version?

ys920826, 2021-03-17:
@liu52020 Hello, has your problem been solved? I also encountered a similar problem. When running in docker, the following error is prompted: "vtkxopen GL render window (0x6da3eb10): cannot create GLX context."

closed 2022-03-08 by github-actions[bot]