open-mmlab/OpenPCDet #400: Failed: To display results of demo.py with Pycharm connecting to a remote ubuntu server

Xelawk, 2020-12-09:
Environment: Pycharm with remote ssh interpreter
Error happens when trying to run demo.py to display the results:
```sh
Got keys from plugin meta data ("xcb")
QFactoryLoader::QFactoryLoader() checking directory path "/home/anaconda3/envs/dev/bin/platforms" ...
loaded library "/home/anaconda3/envs/dev/lib/python3.6/site-packages/PyQt5/Qt/plugins/platforms/libqxcb.so"
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.
```
Does it exists some other ways to display the results, or just saving the results to jpg, png or any other ways?

jihanyang, 2020-12-10:
You can save the point cloud and predicted boxes, and then use mayavi to visualize the result on your local pc or laptop.

Xelawk, 2020-12-10:
> You can save the point cloud and predicted boxes, and then use mayavi to visualize the result on your local pc or laptop.

Thanks

ys920826, 2021-03-17:
@jihanyang hi, thank you for sharing. This is my first contact with mayavi. I also encountered visualization errors when running demo under docker. Can you explain in detail how to modify the code to save the point cloud and predicted boxes. If you can, please explain how to use mayavi on your local PC to load and display. Thanks!

bluestyle97, 2021-04-21:
@jihanyang I save the predicted data and try to visualize it on my macbook, but I the mayavi window shows nothing.
[image]

jihanyang, 2021-04-21:
@bluestyle97 Hello, please provide more details about your problem. I have tried to visualize with mayavi on macbook before, but I haven't meet questions.

closed 2021-05-06 by Xelawk