open-mmlab/OpenPCDet #401: failed to multi-GPU training

arxidinakbar, 2020-12-09:
Hi, all. I tried to train pv_rcnn model with multi-gpu condition, but I failed followed with the command lines in README(not sure if I follow the instructions correctly)
here is the command line:
`bash scripts/dist_train.sh 2 --cfg_file ./cfgs/kitti_models/pv_rcnn.yaml --batch_size 14 --epoch 80 --extra_tag train_Dec`
and got OOM issue, it seems like my command line tried to put all memory footprint in single GPU device where I set 2 GPU for available, I don't know where did I go wrong.
```
Traceback (most recent call last):                                                                                                                                                  | 0/464 [00:00<?, ?it/s]
  File "train.py", line 198, in <module>
    main()
  File "train.py", line 170, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/omnisky/aixiding/OpenPCDet/tools/train_utils/train_utils.py", line 93, in train_model
    dataloader_iter=dataloader_iter
  File "/home/omnisky/aixiding/OpenPCDet/tools/train_utils/train_utils.py", line 38, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/home/omnisky/aixiding/OpenPCDet/pcdet/models/__init__.py", line 30, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/nn/parallel/distributed.py", line 447, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/omnisky/aixiding/OpenPCDet/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/omnisky/aixiding/OpenPCDet/pcdet/models/backbones_2d/base_bev_backbone.py", line 93, in forward
    x = self.blocks[i](x)
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/nn/modules/container.py", line 100, in forward
    input = module(input)
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/nn/modules/conv.py", line 345, in forward
    return self.conv2d_forward(input, self.weight)
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/nn/modules/conv.py", line 342, in conv2d_forward
    self.padding, self.dilation, self.groups)
RuntimeError: CUDA out of memory. Tried to allocate 70.00 MiB (GPU 0; 10.76 GiB total capacity; 9.83 GiB already allocated; 20.19 MiB free; 9.95 GiB reserved in total by PyTorch)
                                                                                                                                                                                                           Traceback (most recent call last):                                                                                                                                                                           
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/distributed/launch.py", line 263, in <module>
    main()
  File "/home/omnisky/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/distributed/launch.py", line 259, in main
    cmd=cmd)
subprocess.CalledProcessError: Command '['/home/omnisky/anaconda3/envs/spconv/bin/python', '-u', 'train.py', '--local_rank=1', '--launcher', 'pytorch', '--cfg_file', './cfgs/kitti_models/pv_rcnn.yaml', '--batch_size', '8', '--extra_tag', 'train_Dec']' returned non-zero exit status 1.
```

I also tried with:
`CUDA_VISIVLE_DEVICES=1,2 python -m torch.distributed.launch --nproc_per_node=2 train.py --launcher pytorch --cfg_file ./cfgs/kitti_models/pv_rcnn.yaml --batch_size 8 --extra_tag train_Dec`
in #74 occurs same issue.
### 
My GPU condition:
2080Ti with 11G memory  *2
```
Want some clue here, please.  Thank you.

arxidinakbar, 2020-12-09:
I find where the problem is, It's because of the batch_size setting too big. Solved the problem by set batch_size=2.
when training, pv_rcnn model occupied about 8700-9100M memory usage on each GPU with batch_size=2.
So, its a rookie mistake.

closed 2020-12-09 by arxidinakbar

direcf, 2021-02-22:
+ same page