open-mmlab/OpenPCDet #402: training pv-rcnn on other dataset

yustaub, 2020-12-09:
nice work firstly. pv-rcnn is an excellent 3D detection network, and I found that it has supported kitti and waymo, and I want to train to it on nuscenes, how can I do this, Can you give me some advices? Very appreciate for your reply!!

yustaub, 2020-12-11:
Can I use the nuscenes data info produced by command `python -m pcdet.datasets.nuscenes.nuscenes_dataset --func create_nuscenes_infos --cfg_file tools/cfgs/dataset_configs/nuscenes_dataset.yaml --version v1.0-trainval` and change the code of kitti_dataset. ____getitem____() to use pv-rcnn.ymal to train nuscenes dataset? @sshaoshuai

yangnvzi, 2021-02-01:
@yustaub 
Have you tried?

yustaub, 2021-02-01:
sorry, I have tried and failed. Maybe there are some wrong in my operarion, but I think it should work

jihanyang, 2021-02-02:
I think it should work. We already have some practice on applying pvrcnn on nuscenes. What is the problem you meet?

yustaub, 2021-02-02:
I tried to transform nuscenes data from nuscenes coordinate system to kitti coordinate system and send it to pvrcnn network to train, and the result is so bad, so I think there must be some wrong in my operation, and when will you release the config of pvrcnn on nuscenes? @jihanyang

yangnvzi, 2021-02-02:
@jihanyang I already have some practice on applying pvrcnn on nuscenes,  but I do by using export_kitti.py, only front camera fov 3dbbox .Do you do the same? or Try to convert to Waymo is worked?

jihanyang, 2021-02-03:
No. We already provide nuscenes dataset.

jihanyang, 2021-02-03:
@yustaub Hello. We directly utilize our nuscenes dataset to train. We find that 2-stage frameworks might not obtain promissing performance on nuscenes.

yangnvzi, 2021-02-04:
@jihanyang 
Do you only use front camera fov 3dbbox ?

yangnvzi, 2021-02-04:
> No. We already provide nuscenes dataset.

what do you mean train to pv-rcnn on nuscenes that directly use nuscenes dataformat?

closed 2022-03-08 by github-actions[bot]

xinshuoweng, 2022-03-08:
@jihanyang Hey Jihan, any chance that I can learn more from you about this (i.e., two-stage detectors like PV-RCNN do not work well on nuScenes)? Any more intuition or observation would be greatly helpful! Also, is the performance still lower than CenterPoint or getting better now?

d33dler, 2022-08-21:
> @jihanyang Hey Jihan, any chance that I can learn more from you about this (i.e., two-stage detectors like PV-RCNN do not work well on nuScenes)? Any more intuition or observation would be greatly helpful! Also, is the performance still lower than CenterPoint or getting better now?
also interested in this, any insight?