open-mmlab/OpenPCDet #403: Failed: undefined symbol

Xelawk, 2020-12-11:
How should I solve this issue
```
Traceback (most recent call last):
  File "test.py", line 9, in <module>
    from pcdet.datasets.my_custom_dataset import MyCustomDataset
  File "/Dev/xelawk_dev/OpenPCDet/pcdet/datasets/__init__.py", line 7, in <module>
    from .dataset import DatasetTemplate
  File "/Dev/xelawk_dev/OpenPCDet/pcdet/datasets/dataset.py", line 8, in <module>
    from .augmentor.data_augmentor import DataAugmentor
  File "/Dev/xelawk_dev/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 6, in <module>
    from . import augmentor_utils, database_sampler
  File "/Dev/xelawk_dev/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 5, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/Dev/xelawk_dev/OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
    from . import iou3d_nms_cuda
ImportError: /Dev/xelawk_dev/OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6caffe28TypeMeta21_typeMetaDataInstanceIN3c108BFloat16EEEPKNS_6detail12TypeMetaDataEv
```

jihanyang, 2020-12-12:
Have your successfully installed spconv and setup pcdet?

Xelawk, 2020-12-14:
> Have your successfully installed spconv and setup pcdet?

My fault, due to the wrong path installing

closed 2020-12-14 by jihanyang

RonaldYuren, 2022-01-25:
> > > > Have your successfully installed spconv and setup pcdet?
> > > 
> > > 
> > > My fault, due to the wrong path installing
> > 
> > 
> > @Xelawk Hi, I'm experiencing the same issue as you. Can you explain what "wrong path installing" is in detail?
> 
> I found it. The path of my model to load is wrong.

sorry, I can't get the point of the   correct "path", what kind of wrong will be?