open-mmlab/OpenPCDet #405: ModuleNotFoundError: No module named 'torch'

hedaniel7, 2020-12-11:
Hello,

I try to use your library for my Bachelor Thesis

When I run ` sudo python3 setup.py develop` in my conda environment this pops up:

```
(env_pytorch) dan@apollo:~/Documents/OpenPCDet$
Traceback (most recent call last):
  File "setup.py", line 5, in <module>
    from torch.utils.cpp_extension import BuildExtension, CUDAExtension
ModuleNotFoundError: No module named 'torch'
```

I have torch installed though and I can import the above modules
```
(env_pytorch) dan@apollo:~/Documents/OpenPCDet$ python
Python 3.6.12 |Anaconda, Inc.| (default, Sep  8 2020, 23:10:56) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> import torchvision
>>> from torch.utils.cpp_extension import BuildExtension, CUDAExtension
>>> exit()
(env_pytorch) dan@apollo:~/Documents/OpenPCDet$ which python
/home/dan/anaconda3/envs/env_pytorch/bin/python
```
I have installed all requirements:
```
(env_pytorch) dan@apollo:~/Documents/OpenPCDet$ pip3 install -r requirements.txt 
Requirement already satisfied: numpy in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (1.19.4)
Requirement already satisfied: torch>=1.1 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 2)) (1.7.1)
Requirement already satisfied: numba in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 3)) (0.52.0)
Requirement already satisfied: tensorboardX in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 4)) (2.1)
Requirement already satisfied: easydict in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 5)) (1.9)
Requirement already satisfied: pyyaml in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 6)) (5.3.1)
Requirement already satisfied: scikit-image in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 7)) (0.17.2)
Requirement already satisfied: tqdm in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 8)) (4.54.1)
Requirement already satisfied: numpy in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (1.19.4)
Requirement already satisfied: llvmlite<0.36,>=0.35.0 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from numba->-r requirements.txt (line 3)) (0.35.0)
Requirement already satisfied: setuptools in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from numba->-r requirements.txt (line 3)) (51.0.0.post20201207)
Requirement already satisfied: tifffile>=2019.7.26 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from scikit-image->-r requirements.txt (line 7)) (2020.9.3)
Requirement already satisfied: matplotlib!=3.0.0,>=2.0.0 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from scikit-image->-r requirements.txt (line 7)) (3.3.3)
Requirement already satisfied: scipy>=1.0.1 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from scikit-image->-r requirements.txt (line 7)) (1.5.4)
Requirement already satisfied: PyWavelets>=1.1.1 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from scikit-image->-r requirements.txt (line 7)) (1.1.1)
Requirement already satisfied: pillow!=7.1.0,!=7.1.1,>=4.3.0 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from scikit-image->-r requirements.txt (line 7)) (8.0.1)
Requirement already satisfied: networkx>=2.0 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from scikit-image->-r requirements.txt (line 7)) (2.5)
Requirement already satisfied: imageio>=2.3.0 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from scikit-image->-r requirements.txt (line 7)) (2.9.0)
Requirement already satisfied: numpy in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (1.19.4)
Requirement already satisfied: six in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from tensorboardX->-r requirements.txt (line 4)) (1.15.0)
Requirement already satisfied: protobuf>=3.8.0 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from tensorboardX->-r requirements.txt (line 4)) (3.14.0)
Requirement already satisfied: numpy in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (1.19.4)
Requirement already satisfied: typing-extensions in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from torch>=1.1->-r requirements.txt (line 2)) (3.7.4.3)
Requirement already satisfied: dataclasses in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from torch>=1.1->-r requirements.txt (line 2)) (0.8)
Requirement already satisfied: numpy in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (1.19.4)
Requirement already satisfied: pillow!=7.1.0,!=7.1.1,>=4.3.0 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from scikit-image->-r requirements.txt (line 7)) (8.0.1)
Requirement already satisfied: numpy in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (1.19.4)
Requirement already satisfied: pillow!=7.1.0,!=7.1.1,>=4.3.0 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from scikit-image->-r requirements.txt (line 7)) (8.0.1)
Requirement already satisfied: numpy in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (1.19.4)
Requirement already satisfied: cycler>=0.10 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from matplotlib!=3.0.0,>=2.0.0->scikit-image->-r requirements.txt (line 7)) (0.10.0)
Requirement already satisfied: kiwisolver>=1.0.1 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from matplotlib!=3.0.0,>=2.0.0->scikit-image->-r requirements.txt (line 7)) (1.3.1)
Requirement already satisfied: python-dateutil>=2.1 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from matplotlib!=3.0.0,>=2.0.0->scikit-image->-r requirements.txt (line 7)) (2.8.1)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.3 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from matplotlib!=3.0.0,>=2.0.0->scikit-image->-r requirements.txt (line 7)) (2.4.7)
Requirement already satisfied: six in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from tensorboardX->-r requirements.txt (line 4)) (1.15.0)
Requirement already satisfied: decorator>=4.3.0 in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from networkx>=2.0->scikit-image->-r requirements.txt (line 7)) (4.4.2)
Requirement already satisfied: six in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from tensorboardX->-r requirements.txt (line 4)) (1.15.0)
Requirement already satisfied: six in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from tensorboardX->-r requirements.txt (line 4)) (1.15.0)
Requirement already satisfied: numpy in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (1.19.4)
Requirement already satisfied: numpy in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (1.19.4)
Requirement already satisfied: numpy in /home/dan/anaconda3/envs/env_pytorch/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (1.19.4)
[... 7 lines cut ...]
| NVIDIA-SMI 450.80.02    Driver Version: 450.80.02    CUDA Version: 11.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce RTX 2070    Off  | 00000000:01:00.0  On |                  N/A |
|  0%   34C    P8     5W / 175W |    180MiB /  7981MiB |      3%      Default |
|                               |                      |                  N/A |
[... cut ...]
|=============================================================================|
|    0   N/A  N/A      1107      G   /usr/lib/xorg/Xorg                 64MiB |
|    0   N/A  N/A      1495      G   /usr/bin/gnome-shell               86MiB |
|    0   N/A  N/A      1996      G   ...token=4170610791772182387       23MiB |
|    0   N/A  N/A      2725      G   /usr/lib/firefox/firefox            2MiB |
+-----------------------------------------------------------------------------+
```
Can you please help me?

Thank you for your time.

jihanyang, 2020-12-12:
Try to use `python setup.py develop` rather than `python3`

HuaxuDuan, 2020-12-14:
env: Ubuntu 18.04 + Python3.6 + PyTorch 1.5 + cuda10.2 + spconv1.2.1
run demo.py
OSError: libcuhash.so: cannot open shared object file: No such file or directory

Daniel-SBR, 2021-02-01:
I had the same problem that seems to be caused by the `sudo` in `sudo python3 setup.py develop`.

Miss out `sudo` and use `python3 setup.py develop --user` worked for me.

stempliu, 2021-11-19:
> I had the same problem that seems to be caused by the `sudo` in `sudo python3 setup.py develop`.
> 
> Miss out `sudo` and use `python3 setup.py develop --user` worked for me.

Thank you so much. It's working for me too.

closed 2022-03-08 by github-actions[bot]

YoucanBaby, 2025-07-26:
Use `sudo python setup.py develop` worked for me.