open-mmlab/OpenPCDet #406: could I get the detection results on image at the same time

arxidinakbar, 2020-12-13:
Hi, all. I've tried demo.py and could get the detection results on point cloud, and wondering is there a simple code for projecting the detection results on image at the same time.
I've tried the codes mentioned here #147, and it didn't work well. So, would appreciate so much if anyone offer some useful tools.

jihanyang, 2020-12-14:
Off course, just project the 3D box into 2D as https://github.com/open-mmlab/OpenPCDet/blob/a7cf5368d9cbc3969b4613c9e61ba4dcaf217517/pcdet/datasets/kitti/kitti_dataset.py#L281-L292.

arxidinakbar, 2020-12-15:
@jihanyang Thank you, this would be helpful! I think I could write couple of code lines to try.

closed 2020-12-27 by jihanyang

jashmehta20, 2021-11-26:
@arxidinakbar Did you manage to visualize the prediction in 2d? I'm stuck and I dont know where to use the above lines od code as they are already written. Thanks for your help!