open-mmlab/OpenPCDet #407: A mistake in `kitti_object_eval_python`

danjia21, 2020-12-13:
Thank you for the great work and for releasing your code! I've spot a mistake in `kitti_object_eval_python`. On the following line, the order of `gt_annos` and `dt_annos` should be swapped. 

https://github.com/open-mmlab/OpenPCDet/blob/a7cf5368d9cbc3969b4613c9e61ba4dcaf217517/pcdet/datasets/kitti/kitti_object_eval_python/eval.py#L473

The (i, j) index on this line should also be swapped consistently.
 
https://github.com/open-mmlab/OpenPCDet/blob/a7cf5368d9cbc3969b4613c9e61ba4dcaf217517/pcdet/datasets/kitti/kitti_object_eval_python/eval.py#L207

TarunKumar1995-glitch, 2021-10-05:
https://github.com/open-mmlab/OpenPCDet/issues/139

This is already answered and closed in another issue. You might want to give it a check.

danjia21, 2021-10-05:
Thanks for the pointer.