open-mmlab/OpenPCDet #411: Confusion about convertion rotation_y in kitti

yustaub, 2020-12-20:
Hello,
    I am confused about rot_y convertion between kitti image plane and kitti lidar coordinate system, [lidar_to_image](https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/utils/box_utils.py#L165), [image_to_lidar](https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/kitti/kitti_dataset.py#L166), I know that their coordinate system is different(image: x right, y down, z front; lidar: x front, y left, z up), can you explain that how the convertion works? Very appreciate for your reply!

closed 2022-03-08 by github-actions[bot]

himanshuak, 2022-07-26:
Hello, Did you by any chance get this one. I am trying to get the pointcloud data and ground using Carla Simulator. But I am confused about the sign convention for rotation_y. Any suggestions would be helpful

WinckelData, 2022-11-24:
DId you find anything out? @himanshuak