open-mmlab/OpenPCDet #415: all the AP is 0 when training on waymo dataset

duanchengwen, 2020-12-24:
here is the data i used
├── training_0000.tar
├── training_0001.tar
├── training_0002.tar
├── validation_0000.tar
├── validation_0001.tar

the training this data on second model with the latest codes. but all the accuracy is 0

`2020-12-22 21:01:49,072   INFO  **********************Start logging**********************
2020-12-22 21:01:49,072   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-12-22 21:01:49,072   INFO  cfg_file         cfgs/waymo_models/second.yaml
2020-12-22 21:01:49,072   INFO  batch_size       1
2020-12-22 21:01:49,072   INFO  epochs           30
2020-12-22 21:01:49,072   INFO  workers          0
2020-12-22 21:01:49,072   INFO  extra_tag        default
2020-12-22 21:01:49,072   INFO  ckpt             None
2020-12-22 21:01:49,073   INFO  pretrained_model None
2020-12-22 21:01:49,073   INFO  launcher         none
2020-12-22 21:01:49,073   INFO  tcp_port         18888
2020-12-22 21:01:49,073   INFO  sync_bn          False
2020-12-22 21:01:49,073   INFO  fix_random_seed  False
2020-12-22 21:01:49,073   INFO  ckpt_save_interval 1
2020-12-22 21:01:49,073   INFO  local_rank       0
2020-12-22 21:01:49,073   INFO  max_ckpt_save_num 30
2020-12-22 21:01:49,073   INFO  merge_all_iters_to_one_epoch False
2020-12-22 21:01:49,073   INFO  set_cfgs         None
2020-12-22 21:01:49,073   INFO  max_waiting_mins 0
2020-12-22 21:01:49,073   INFO  start_epoch      0
2020-12-22 21:01:49,073   INFO  save_to_file     False
2020-12-22 21:01:49,073   INFO  cfg.ROOT_DIR: /home/zombie/zombie/deepleaning/lidar/OpenPCDet
2020-12-22 21:01:49,073   INFO  cfg.LOCAL_RANK: 0
2020-12-22 21:01:49,073   INFO  cfg.CLASS_NAMES: ['Vehicle', 'Pedestrian', 'Cyclist']
2020-12-22 21:01:49,073   INFO  
cfg.DATA_CONFIG = edict()
2020-12-22 21:01:49,073   INFO  cfg.DATA_CONFIG.DATASET: WaymoDataset
2020-12-22 21:01:49,073   INFO  cfg.DATA_CONFIG.DATA_PATH: /media/zombie/DATA/waymo_dataset
2020-12-22 21:01:49,073   INFO  cfg.DATA_CONFIG.PROCESSED_DATA_TAG: waymo_processed_data
2020-12-22 21:01:49,073   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [-75.2, -75.2, -2, 75.2, 75.2, 4]
2020-12-22 21:01:49,073   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2020-12-22 21:01:49,073   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2020-12-22 21:01:49,074   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2020-12-22 21:01:49,074   INFO  
cfg.DATA_CONFIG.SAMPLED_INTERVAL = edict()
2020-12-22 21:01:49,074   INFO  cfg.DATA_CONFIG.SAMPLED_INTERVAL.train: 5
2020-12-22 21:01:49,074   INFO  cfg.DATA_CONFIG.SAMPLED_INTERVAL.test: 5
2020-12-22 21:01:49,074   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2020-12-22 21:01:49,074   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2020-12-22 21:01:49,074   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['pcdet_waymo_dbinfos_train_sampled_10.pkl'], 'PREPARE': {'filter_by_min_points': ['Vehicle:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Vehicle:15', 'Pedestrian:10', 'Cyclist:10'], 'NUM_POINT_FEATURES': 5, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': True}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x', 'y']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2020-12-22 21:01:49,074   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2020-12-22 21:01:49,074   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2020-12-22 21:01:49,074   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity', 'elongation']
2020-12-22 21:01:49,074   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity', 'elongation']
2020-12-22 21:01:49,074   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': True}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.1, 0.1, 0.15], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 80000, 'test': 90000}}]
2020-12-22 21:01:49,074   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: /home/zombie/zombie/deepleaning/lidar/OpenPCDet/tools/cfgs/dataset_configs/waymo_dataset.yaml
2020-12-22 21:01:49,074   INFO  
cfg.MODEL = edict()
2020-12-22 21:01:49,074   INFO  cfg.MODEL.NAME: SECONDNet
2020-12-22 21:01:49,074   INFO  
cfg.MODEL.VFE = edict()
2020-12-22 21:01:49,074   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2020-12-22 21:01:49,074   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2020-12-22 21:01:49,074   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2020-12-22 21:01:49,074   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2020-12-22 21:01:49,074   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
[... 314 lines cut ...]
2020-12-23 10:39:42.324978: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:159] Parsing prediction [115125,7][115125]
2020-12-23 10:39:42.350499: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:168] Parsing ground truth [0,7][0]
2020-12-23 10:39:42.472671: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:214] Done with computing detection metrics.
2020-12-23 10:39:42,531   INFO  
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.0000 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.0000 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.0000 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.0000 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.0000 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.0000 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.0000 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.0000 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.0000 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.0000 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.0000 

2020-12-23 10:39:42,531   INFO  Result is save to /home/zombie/zombie/deepleaning/lidar/OpenPCDet/output/waymo_models/second/default/eval/eval_with_train/epoch_30/val
2020-12-23 10:39:42,531   INFO  ****************Evaluation done.*****************
2020-12-23 10:39:42,538   INFO  Epoch 30 has been evaluated
2020-12-23 10:40:12,565   INFO  **********************End evaluation waymo_models/second(default)**********************o_models/second/default/ckpt `

JingyaLiu, 2021-02-22:
I have the same issue. Did you solve it?

duanchengwen, 2021-03-03:
> I have the same issue. Did you solve it?

have not

xuxuxuchen, 2021-04-22:
I have the same issue, too. Did you solve it?

YongYongLi123, 2021-10-16:
I have the same issue, too. Did you solve it?

jihanyang, 2022-02-11:
Notice that the number of gt is 0.
```
2020-12-23 10:39:38,760 INFO Average predicted number of objects(1983 samples): 58.056
Start the waymo evaluation...
Number: (pd, 115125) VS. (gt, 0)
```

Viczyf, 2022-02-11:
num_points_in_gt.append(laser_labels[i].num_lidar_points_in_box)

I dont' know why all num_points_in_gt is 0 in waymo_processed_data_v0_5_0_infos_train.pkl.

Viczyf, 2022-02-11:
def generate_labels(frame):
    obj_name, difficulty, dimensions, locations, heading_angles = [], [], [], [], []
    tracking_difficulty, speeds, accelerations, obj_ids = [], [], [], []
    num_points_in_gt = []
    laser_labels = frame.laser_labels
    for i in range(len(laser_labels)):
        box = laser_labels[i].box
        class_ind = laser_labels[i].type
        loc = [box.center_x, box.center_y, box.center_z]
        heading_angles.append(box.heading)
        obj_name.append(WAYMO_CLASSES[class_ind])
        difficulty.append(laser_labels[i].detection_difficulty_level)
        tracking_difficulty.append(laser_labels[i].tracking_difficulty_level)
        dimensions.append([box.length, box.width, box.height])  # lwh in unified coordinate of OpenPCDet
        locations.append(loc)
        obj_ids.append(laser_labels[i].id)
        num_points_in_gt.append(laser_labels[i].num_lidar_points_in_box)

I mean num_lidar_points_in_box in  frame.laser_labels is 0. So in raw data the num_lidar_points_in_box is 0;
Can you please test the code? @jihanyang

closed 2022-03-28 by github-actions[bot]

carrolltm, 2023-05-25:
> > I have the same issue. Did you solve it?
> 
> have not

I have the same issue. Did you solve it?

fx110127, 2024-02-25:
I have the same issue. Did you solve it?

Eaphan, 2024-02-25:
The error happened to me a long time ago. It seems that you should download the Waymo v1.2 dataset instead of v1.0 dataset.