open-mmlab/OpenPCDet #417: There is my waymo dataset model prediction results, I think the ped/cyc length and width are wrong!

PengfeiMa, 2020-12-24:
As describe, but the car bboxes prediction is right, what's wrong with the ped/cyc？ @sshaoshuai 
here are some result and bboxes configs.
 
[image]
[image]
[image]

yrh2526, 2020-12-24:
Have you ever tried training on the kitti dataset?

PengfeiMa, 2020-12-25:
> Have you ever tried training on the kitti dataset?

kitti dataset is OK, but waymo coordinate origin is ground, so are there some error configs in anchor_bottom_heights?

yrh2526, 2020-12-25:
> but waymo coordinate origin is ground, so are there some error configs in anchor_bottom_heights?

From the graph, there are some errors in pedestrian ,but from the anchor size, it is reasonable,could it be an error in visual?
Could you show me your envs you trained the kitti dataset?

PengfeiMa, 2020-12-25:
> > but waymo coordinate origin is ground, so are there some error configs in anchor_bottom_heights?
> 
> From the graph, there are some errors in pedestrian ,but from the anchor size, it is reasonable,could it be an error in visual?
> Could you show me your envs you trained the kitti dataset?

Think you for your reply, I find the error, I used the normalization for the z value in the pointcloud when generate the voxel.
I del the normalize, it looks better.

closed 2021-01-07 by PengfeiMa