open-mmlab/OpenPCDet #423: How to use path with \

oldtime-tai, 2021-01-05:
I have a little question.

When  I read the demo code, I find you use \ to represent the directory path, such as 

output_dir = cfg.ROOT_DIR / 'output' / cfg.EXP_GROUP_PATH / cfg.TAG / args.extra_tag
[image]
 
I suppose this is a good programming method， but i fail with this error.

    output_dir = 'checkpoint' / 'test_1'
TypeError: unsupported operand type(s) for /: 'str' and 'str'
[image]

Can you tell me why?

Tanks

closed 2021-01-05 by oldtime-tai

wangyakunn, 2023-04-04:
I meet the same question.

wangyakunn, 2023-04-04:
Hi，I met the same question, could you tell me how to solve?

thanks